balena-os/meta-balena #2026: Reduce dependence on `/etc/resin-supervisor/supervisor.conf`

xginn8, 2020-10-12:
An empty `/etc/resin-supervisor/supervisor.conf` made the supervisor continuously fail to start. we source that file as an EnvironmentFile for the supervisor, but that file shouldn't be relied on at all if a cloud is available to confirm the target version.

`LED_FILE` information should also be moved to hw contracts or device-level information generally.

ref. https://www.flowdock.com/app/rulemotion/r-supervisor/threads/wRJTK4BukX0glyt_xHtpJD0wwS5

xginn8, 2020-10-12:
ref. https://github.com/balena-os/balenahup/issues/310

alexgg, 2020-11-17:
@xginn8 @CameronDiver we still want to start a supervisor even if the cloud is not accessible, so ideally the system should be able to infer which of the installed images is the supervisor. Something we could do is add label to the supervisor image so that the system can recognize it. That way `start-resin-supervisor` would always run the installed supervisor image, instead of reading the environment to identify the image.

xginn8, 2020-11-19:
ultimately the local fallback source of truth will be `apps.json`, like all other apps, right? this file is vestigial in the sense that there's a generic way to handle this issue.

alexgg, 2020-11-23:
@xginn8 as I understand it, `apps.json` only contains the preloaded state, not the current state. Ideally we would like the supervisor start script to:
* Start the currently installed supervisor, even if there is no cloud connectivity
* If there is no, fallback to `apps.json`

It's point nr 1 that I was referring to above.

klutchell, 2022-01-12:
This file is scheduled to be removed when supervisor becomes a proper host app.

closed 2022-01-12 by klutchell