balena-os/meta-balena #2027: Root /dev/disk/by-label/flash-rootA doesen't exist or does not contain a /dev

sxiii, 2020-10-13:
Greetings.

Trying to boot "generic x86_64" image.

It stopped at the error from title:
```
Root /dev/disk/by-label/flash-rootA doesen't exist or does not contain a /dev
```
Is there anything for me to try or is it a bug?

Thanks.

sxiii, 2020-10-13:
Closing, as this had something to do with "read-write" mode of DriveDroid. I switched it to "Read-Only" mode and it seems that system booted OK.

closed 2020-10-13 by sxiii

splatch, 2022-04-13:
I know its an old issue, but I can put a short comment on the issue. If you see such error it is a result of configuration mismatch.

For me it happened when drive mounted (pointed to kernel) by bootloader was not properly detected and mapped in `/dev` tree. I found your issue while looking for an solution. We got it reproduced with yocto image made for ssd (sata) drive attached via ata controller in vbox. All partitions were fine, but `root=LABEL=rootfsA` kernel parameter (`flash-rootA` for you) was not working. I did a test attaching second filesystem via other controller kind (ide) and labels for partitions such as `hda` are not automatically put in `/dev/disks/by-label`. Not sure why is that and can't debug it further as we fixed configuration error we found, so leaving this comment here for others who will look for "doesen't exist or does not contain a /dev". :)