balena-os/meta-balena #2039: u-blox-mode-switch script failing frequently

markcorbinuk, 2020-10-30:
During testing for issue #1962 it was found that the u-blox-switch script is only successful 25% of the time (or less). The script remains running continuously and blocks network startup (even when an alternative connection method is available, e.g. WiFi).

Testing was conducted using a balena Fin with a u-blox Toby L200 modem. A test script was written to revert the modem to RNDIS mode. Once in RNDIS mode the system was rebooted and monitored via a serial link to determine whether u-blox-switch.sh had run correctly. Minicom was also deployed to the target to allow direct interaction with the modem device for debugging purposes.

acostach, 2021-03-03:
Leaving a note here for when this ticket will be handled, the reboot triggered from the u-blox-switch.sh script may cause issues if it happened right after an OS update, before the rollback scripts clear the breadcrumbs. Did not reproduce this issue, but it needs to be taken into account.

floion, 2021-04-20:
Fixed by https://github.com/balena-os/meta-balena/pull/2161

closed 2021-04-20 by floion