balena-os/meta-balena #2044: Improve robustness of time synchronisation at boot

markcorbinuk, 2020-11-03:
At boot time chrony currently sends up to 8 NTP requests at 2 second intervals. Time synchronisation is only achieved if 3 good replies are received during this period. The default NTP poll rate is approximately 4.5 hours, so if the initial polling burst is unsuccessful it will take at least an additional 13.5 hours to achieve synchronisation.

The time synchronisation process at boot needs to be improved for cases where the NTP servers are unreliable or the system has delayed access to internet services.

markcorbinuk, 2020-11-17:
A user has reported a `chrony` synchronisation that could be related to DNS and the sequencing of network services at boot. It looks likely that `chrony` is running the initial burst before it is able to resolve NTP server addresses:
```
Nov 16 20:00:05 localhost chronyd[1307]: 2020-11-16T20:00:05Z chronyd version 3.4 starting (+CMDMON +NTP +REFCLOCK +RTC -PRIVDROP -SCFILTER -SIGND +ASYNCDNS -SECHASH +IPV6 -DEBUG)
Nov 16 20:00:05 localhost chronyd[1307]: 2020-11-16T20:00:05Z Frequency 1.924 +/- 0.493 ppm read from /var/lib/chrony/drift

Nov 16 20:00:08 e7430c0 dnsmasq[1384]: dnsmasq: syntax check OK.
Nov 16 20:00:08 e7430c0 dnsmasq[1389]: dnsmasq[1389]: started, version 2.78 cachesize 150
Nov 16 20:00:08 e7430c0 dnsmasq[1389]: dnsmasq[1389]: compile time options: IPv6 GNU-getopt DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth no-DNSSEC loop-detect inotify
Nov 16 20:00:08 e7430c0 dnsmasq[1389]: dnsmasq[1389]: DBus support enabled: connected to system bus
Nov 16 20:00:08 e7430c0 dnsmasq[1389]: dnsmasq[1389]: reading /etc/resolv.dnsmasq
Nov 16 20:00:08 e7430c0 dnsmasq[1389]: dnsmasq[1389]: using nameserver 8.8.8.8#53
Nov 16 20:00:08 e7430c0 dnsmasq[1389]: dnsmasq[1389]: read /etc/hosts - 6 addresses
Nov 16 20:00:08 e7430c0 dnsmasq[1389]: dnsmasq[1389]: using nameserver 8.8.8.8#53

<Guessing that initial chrony burst finished at 20:00:21?? >

Nov 16 20:00:54 e7430c0 dnsmasq[1389]: dnsmasq[1389]: reading /etc/resolv.dnsmasq
Nov 16 20:00:54 e7430c0 dnsmasq[1389]: dnsmasq[1389]: using nameserver 8.8.8.8#53
Nov 16 20:00:54 e7430c0 dnsmasq[1389]: dnsmasq[1389]: using nameserver 192.168.0.1#53

Nov 16 20:01:24 e7430c0 dnsmasq[1389]: dnsmasq[1389]: reading /etc/resolv.dnsmasq
Nov 16 20:01:24 e7430c0 dnsmasq[1389]: dnsmasq[1389]: using nameserver 8.8.8.8#53
Nov 16 20:01:24 e7430c0 dnsmasq[1389]: dnsmasq[1389]: using nameserver 193.162.153.164#53
Nov 16 20:01:24 e7430c0 dnsmasq[1389]: dnsmasq[1389]: using nameserver 194.239.134.83#53
<Only configured chrony source at this point was 192.168.0.1>

Nov 17 00:36:56 e7430c0 chronyd[1307]: 2020-11-17T00:36:56Z Source 192.168.0.1 replaced with 194.192.112.20
```

markcorbinuk, 2020-11-19:
Upon further testing I believe that the DNS related scenario shown above is due to network access via a captive portal (or similar). Chrony only manages to resolve a single IP address (192.168.0.1) on startup which was shown as `_gateway` in the output of `chronyc sources`. The sources were eventually refreshed 4.5 hours later and 192.168.0.1 was replaced with 194.192.112.20.

I can reproduce the above behaviour by configuring `dnsmasq` to return a single address (my network gateway) for all DNS lookups. Chrony tries to resolve `0.resinio.pool.ntp.org`, `1.resinio.pool.ntp.org`, `2.resinio.pool.ntp.org` and `3.resinio.pool.ntp.org`, but they all return the same IP (192.168.12.1 in this case) so chrony configures a single source using this address. Chrony then sends an NTP burst to the this IP address, but as it is not an NTP server no replies are received and the source is marked unreachable. When DNS is finally configured correctly chrony eventually refreshes the NTP sources and gets a valid NTP server address. However because only one source has been configured initially and a source refresh only updates the IP addresses of existing sources, the system continues running with a single NTP server instead of 4.

@majorz Does this make sense to you? Do you think that we need to implement captive portal detection by default? I see that Android uses http://clients3.google.com/generate_204 or http://connectivitycheck.android.com/generate_204 to try and detect captive portals ( as referenced here: https://github.com/balena-io/wifi-connect/issues/51 ).

anathan, 2020-11-19:
I think the captive portals bus likely an issue for some, but we've seen this on setups we know do not have a captive portal. The WiFi is a Pepwave router on T-Mobile and that device will check for internet health before providing any external routes to its clients.

markcorbinuk, 2020-11-19:
>...that device will check for internet health before providing any external routes to its clients.

So it looks like a more thorough connectivity check before starting `chrony` would be good in both cases.

majorz, 2020-12-01:
@markcorbinuk I stumbled across the same issue (same customer) and was able to take a look at the device logs.

The device was connected through Ethernet (eth0) to the Internet (not WiFi). For half a day after the device was first booted the router did not provide any Internet connectivity. I saw the same symptoms with DNS misconfiguration - our VPN being resolved to 192.168.0.1.

At one point the device regained Internet connectivity:
```
Nov 30 17:26:25 25d5fd4 NetworkManager[1387]: <info>  [1606757185.2494] manager: NetworkManager state is now CONNECTED_GLOBAL
```

Half an hour later chrony replaced its source:
```
Nov 30 18:13:00 25d5fd4 chronyd[1304]: 2020-11-30T18:13:00Z Source 192.168.0.1 replaced with 195.137.195.254
```

Half a day later the customer saw that time was very incorrect. After he manually restarted chrony the time was synchronized properly this time.
```
Dec 01 04:30:44 25d5fd4 chronyd[1304]: 2020-12-01T04:30:44Z chronyd exiting
Dec 01 04:30:44 25d5fd4 chronyd[7975]: 2020-12-01T04:30:44Z chronyd version 3.4 starting (+CMDMON +NTP +REFCLOCK +RTC -PRIVDROP -SCFILTER -SIGND +ASYNCDNS -SECHASH +IPV6 -DEBUG)
Dec 01 04:30:44 25d5fd4 chronyd[7975]: 2020-12-01T04:30:44Z Frequency 1.899 +/- 0.005 ppm read from /var/lib/chrony/drift
Dec 01 04:30:51 25d5fd4 chronyd[7975]: 2020-12-01T04:30:51Z Selected source 162.159.200.123
Dec 01 07:15:42 25d5fd4 chronyd[7975]: 2020-12-01T04:30:51Z System clock wrong by 9890.431308 seconds, adjustment started
Dec 01 07:15:42 25d5fd4 chronyd[7975]: 2020-12-01T07:15:42Z System clock was stepped by 9890.431308 seconds
Dec 01 07:15:42 25d5fd4 fake-hwclock[7981]: [fake-hwclock][INFO] Saving system time to /etc/fake-hwclock/fake-hwclock.data.
```

Even though the router was misbehaving at the initial state, the manual chrony restart fixed the time synchronization later on when Internet connectivity was available. It looks like there is definitely room for improvement on our side and manual intervention should not be needed.

alexgg, 2021-03-19:
@markcorbinuk this seems too generic for an issue. With the work done so far, could we rename it so it reflects concrete pending work?

minusplusminus, 2021-09-10:
Hi I had an issue similar to this. One of my devices ( balenaOS 2.47.0+rev1 supervisor 10.6.27)  was offline  for 19 hours.  Suddenly popped online again. It uses  ModemManager to create a LTE connection. Does updating OS solve the issue or am I then looking the wrong way? 
```
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6521] modem["cdc-wdm0"]: modem state changed, 'connecting' --> 'connected' (reason: user-requested)
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6686] device (cdc-wdm0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6728] device (cdc-wdm0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6763] modem-broadband[cdc-wdm0]: IPv4 static configuration:
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6772] modem-broadband[cdc-wdm0]:   address 10.18.21.142/30
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6778] modem-broadband[cdc-wdm0]:   gateway 10.18.21.141
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6782] modem-broadband[cdc-wdm0]:   DNS 194.151.228.2
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6788] modem-broadband[cdc-wdm0]:   DNS 194.151.228.34
Sep 09 23:07:54 neonlink 5d3fd0d2295e[785]: [185B blob data]
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.6849] device (cdc-wdm0): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
Sep 09 23:07:54 neonlink resin-supervisor[1247]: resin_supervisor
Sep 09 23:07:54 neonlink 5d3fd0d2295e[785]: [149B blob data]
Sep 09 23:07:54 neonlink 5d3fd0d2295e[785]: [57B blob data]
Sep 09 23:07:54 neonlink resin-supervisor[1587]: active
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.7620] device (cdc-wdm0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.7637] device (cdc-wdm0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.7669] manager: NetworkManager state is now CONNECTED_LOCAL
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.7808] manager: NetworkManager state is now CONNECTED_SITE
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.7837] policy: set 'cellular' (wwan0) as default for IPv4 routing and DNS
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.7901] dns-mgr: Writing DNS information to /sbin/resolvconf
Sep 09 23:07:54 neonlink dnsmasq[782]: dnsmasq[782]: reading /etc/resolv.dnsmasq
Sep 09 23:07:54 neonlink dnsmasq[782]: dnsmasq[782]: using nameserver 8.8.8.8#53
Sep 09 23:07:54 neonlink dnsmasq[782]: dnsmasq[782]: using nameserver 194.151.228.2#53
Sep 09 23:07:54 neonlink dnsmasq[782]: dnsmasq[782]: using nameserver 194.151.228.34#53
Sep 09 23:07:54 neonlink NetworkManager[773]: <info>  [1631228874.8462] device (cdc-wdm0): Activation: successful, device activated.
Sep 09 23:07:55 neonlink nm-dispatcher[1588]: resin-ntp-config: Found config.json in /mnt/boot/config.json .
Sep 09 23:07:55 neonlink nm-dispatcher[1588]: 200 OK
Sep 09 23:07:55 neonlink NetworkManager[773]: <info>  [1631228875.4033] manager: NetworkManager state is now CONNECTED_GLOBAL
Sep 09 23:07:55 neonlink nm-dispatcher[1588]: resin-ntp-config: Found config.json in /mnt/boot/config.json .
Sep 09 23:07:55 neonlink nm-dispatcher[1588]: 200 OK
Sep 09 23:07:58 neonlink resin-supervisor[1589]: Container config has not changed
Sep 09 23:07:59 neonlink balenad[785]: time="2021-09-09T23:07:59.100593990Z" level=info msg="shim balena-engine-containerd-shim started" address=/containerd-shim/moby/0ba84a7eed488c51e4672d13185057df9364d3ff0281cbbbc13174f8a489f97f/shim.sock debug=false pid=1703
Sep 09 23:08:00 neonlink resin-supervisor[1589]: Starting system message bus: dbus.
Sep 09 23:08:00 neonlink 0ba84a7eed48[785]: Starting system message bus: dbus.
Sep 09 23:08:00 neonlink 0ba84a7eed48[785]:  * Starting Avahi mDNS/DNS-SD Daemon: avahi-daemon
Sep 09 23:08:00 neonlink resin-supervisor[1589]:  * Starting Avahi mDNS/DNS-SD Daemon: avahi-daemon
Sep 09 23:08:00 neonlink 0ba84a7eed48[785]:    ...done.
Sep 09 23:08:00 neonlink resin-supervisor[1589]:    ...done.
Sep 09 23:08:00 neonlink chronyd[723]: 2021-09-09T23:08:00Z Selected source 5.79.108.34
Sep 09 23:08:00 neonlink chronyd[723]: 2021-09-09T23:08:00Z System clock wrong by 68371.484508 seconds, adjustment started
Sep 10 18:07:32 neonlink chronyd[723]: 2021-09-10T18:07:32Z System clock was stepped by 68371.484508 seconds
Sep 10 18:07:32 neonlink journalctl[1772]: Vacuuming done, freed 0B of archived journals from /run/log/journal/574e2a7b0fc64fa190b8725c4bdd5d10.
Sep 10 18:07:33 neonlink chronyd[723]: 2021-09-10T18:07:33Z Selected source 94.198.159.10
Sep 10 18:07:33 neonlink 5d3fd0d2295e[785]: [58B blob data]
Sep 10 18:07:33 neonlink 5d3fd0d2295e[785]: [11B blob data]
Sep 10 18:07:33 neonlink 5d3fd0d2295e[785]: [18B blob data]
Sep 10 18:07:33 neonlink 5d3fd0d2295e[785]: [15B blob data]
Sep 10 18:07:33 neonlink 5d3fd0d2295e[785]: [4B blob data]
Sep 10 18:07:33 neonlink 5d3fd0d2295e[785]: [13B blob data]

```

alexgg, 2021-09-13:
hi @minusplusminus at first glance your problem seems unrelated to this issue. If it's reproducible I would recommend to use the latest available balenaOS release for your device type which includes newer Networkmanager and ModemManager versions and report it through our forums (https://forums.balena.io) so we can help debugging it.

closed 2021-12-01 by bulldozer-balena[bot]