balena-os/meta-balena #2049: update-resin-supervisor seems to be restarting supervisor without a reason

majorz, 2020-11-12:
Device type: Balena Fin (CM3)
OS version: balenaOS 2.51.1+rev1
Supervisor version: 11.4.10

During a device investigation I stumbled across supervisor being restarted without apparent reason:
```
Nov 09 22:24:41 09b488a systemd-tmpfiles[4172]: [/etc/tmpfiles.d/openvpn.conf:1] Line references path below legacy directory /var/run/, updating /var/run/openvpn → /run/openvpn; please update the tmpfiles.d/ drop-in file accordingly.
Nov 09 22:24:41 09b488a sh[4171]: Getting image name and tag...
Nov 09 22:24:42 09b488a sh[4171]: Supervisor configuration found from API.
Nov 09 22:24:42 09b488a sh[4171]: Stop supervisor...
```

I omit the previous logs as they are from much earlier time and are unrelated. The first line is a strange warning issued by the time `update-resin-supervisor` was executed.

We need to review the conditions update-resin-supervisor restarts the supervisor.

alexgg, 2020-11-17:
@majorz it seems `update-resin-supervisor` fetches a new supervisor release target from the state endpoint, checks that the image is not present in the device, and stops the supervisor. There must be more log messages to see what happens next, as the script should log a message if the pull failed and always restart the supervisor.
But I don't see a problem with the logic in the script. What happened is that the target state endpoint for this device was updated.

alexgg, 2020-12-15:
After reading the long thread above I don't see a clear indication that there is an OS problem here. I am closing this for now as the thread has been summarized.

closed 2020-12-15 by alexgg