balena-os/meta-balena #2107: Filesystem unmounting problems at shutdown

markcorbinuk, 2021-01-29:
Whilst looking at persistent logging issue #1919 various problems with unmounting filesystems at shutdown have been observed:
```
1  Jan 28 13:27:15 c026ba6 fake-hwclock[2471]: [fake-hwclock][INFO] Saving system time to /etc/fake-hwclock/fake-hwclock.data.
2  Jan 28 13:27:15 c026ba6 fake-hwclock[2481]: /sbin/fake-hwclock: line 52: /etc/fake-hwclock/fake-hwclock.data: Read-only file system
3  Jan 28 13:27:15 c026ba6 umount[2487]: umount: /var/volatile/lib/docker: not mounted.
4  Jan 28 13:27:15 c026ba6 umount[2488]: umount: /var/volatile/log/journal: target is busy.
5  Jan 28 13:27:15 c026ba6 umount[2489]: umount: /var/cache: not mounted.
6  Jan 28 13:27:15 c026ba6 umount[2490]: umount: /var/spool: not mounted.
7  Jan 28 13:27:15 c026ba6 umount[2491]: umount: /srv: not mounted.
8  Jan 28 13:27:15 c026ba6 systemd[1]: fake-hwclock.service: Control process exited, code=exited, status=1/FAILURE
9  Jan 28 13:27:15 c026ba6 systemd[1]: fake-hwclock.service: Failed with result 'exit-code'.
10 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-lib-docker.mount: Mount process exited, code=exited, status=32/n/a
11 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-lib-docker.mount: Failed with result 'exit-code'.
12 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-log-journal.mount: Mount process exited, code=exited, status=32/n/a
13 Jan 28 13:27:16 c026ba6 systemd[1]: Failed unmounting /var/volatile/log/journal.
14 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-cache.service: Control process exited, code=exited, status=32/n/a
15 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-cache.service: Failed with result 'exit-code'.
16 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-spool.service: Control process exited, code=exited, status=32/n/a
17 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-spool.service: Failed with result 'exit-code'.
18 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-srv.service: Control process exited, code=exited, status=32/n/a
19 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-srv.service: Failed with result 'exit-code'.
20 Jan 28 13:27:16 c026ba6 bindmount[2494]: INFO: Unmounting /mnt/state/root-overlay/etc/fake-hwclock ...
21 Jan 28 13:27:16 c026ba6 bindmount[2494]: INFO: bind mountpont is already unmounted.
22 Jan 28 13:27:16 c026ba6 umount[2496]: umount: /var/lib: not mounted.
23 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-lib.service: Control process exited, code=exited, status=32/n/a
24 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile-lib.service: Failed with result 'exit-code'.
25 Jan 28 13:27:16 c026ba6 resin-partition-mounter[2495]: INFO: /mnt/data already unmounted.
26 Jan 28 13:27:16 c026ba6 umount[2499]: umount: /var/volatile: target is busy.
27 Jan 28 13:27:16 c026ba6 systemd[1]: var-volatile.mount: Mount process exited, code=exited, status=32/n/a
28 Jan 28 13:27:16 c026ba6 systemd[1]: Failed unmounting /var/volatile.
29 Jan 28 13:27:16 c026ba6 resin-partition-mounter[2500]: INFO: /mnt/state already unmounted.
30 Jan 28 13:27:16 c026ba6 systemd[1]: Shutting down.
31 Jan 28 13:27:16 c026ba6 systemd[1]: Failed to set timeout to 600s: Invalid argument
32 Jan 28 13:27:16 c026ba6 systemd-journald[911]: Journal stopped
```

Lines 1 & 2 show that something has unmounted `/mnt/state/root-overlay/etc/fake-hwclock` before `fake-hwclock` runs causing it to fail. This can also be seen in lines 20 & 21. This problem only occurs occasionally.

It can also be seen that the system is failing to unmount a variety of other filesystems including `/var/volatile/log/journal` which may be contributing to persistent log problems observed elsewhere.

markcorbinuk, 2021-02-04:
It seems that systemd automatically creates runtime mount units for the filesystems that are mounted by the various balena systemd services. For example running `systemctl` shows that a runtime mount unit called `etc-fake\x2dhwclock.mount` exists for the bindmount created by the `bind-etc-fake-hwclock.service`. Similar mount units are created for other services.

The systemd.mount documentation says `Mount points created at runtime (independently of unit files or /etc/fstab) will be monitored by systemd and appear like any other mount unit in systemd.`

At shutdown there seems to be a clash between the runtime created mount unit and the systemd service with respect to which one successfully unmounts the filesystem first.

The following log shows the `bind-etc-fake-hwclock.service` unmounting the it's filesystem before`etc-fake\x2dhwclock.mount` tries to do the same thing:
```
Feb 04 13:31:56 ff4ffca systemd[1]: bind-etc-fake-hwclock.service: About to execute: /usr/bin/bindmount --target /etc/fake-hwclock --bind-root /mnt/state/root-overlay --command unmo>
Feb 04 13:31:56 ff4ffca systemd[1]: bind-etc-fake-hwclock.service: Forked /usr/bin/bindmount as 3480
Feb 04 13:31:56 ff4ffca systemd[1]: bind-etc-fake-hwclock.service: Changed exited -> stop
Feb 04 13:31:56 ff4ffca systemd[1]: Stopping Bind mount for /etc/fake-hwclock...
Feb 04 13:31:56 ff4ffca systemd[3480]: bind-etc-fake-hwclock.service: Executing: /usr/bin/bindmount --target /etc/fake-hwclock --bind-root /mnt/state/root-overlay --command unmount
Feb 04 13:31:56 ff4ffca systemd[1]: etc-fake\x2dhwclock.mount: About to execute: /bin/umount /etc/fake-hwclock -c
Feb 04 13:31:56 ff4ffca systemd[1]: etc-fake\x2dhwclock.mount: Forked /bin/umount as 3481
Feb 04 13:31:56 ff4ffca systemd[1]: etc-fake\x2dhwclock.mount: Changed mounted -> unmounting
Feb 04 13:31:56 ff4ffca systemd[1]: Unmounting /etc/fake-hwclock...
Feb 04 13:31:56 ff4ffca systemd[3481]: etc-fake\x2dhwclock.mount: Executing: /bin/umount /etc/fake-hwclock -c
Feb 04 13:31:56 ff4ffca bindmount[3480]: INFO: Unmounting /mnt/state/root-overlay/etc/fake-hwclock ...
Feb 04 13:31:56 ff4ffca bindmount[3480]: INFO: Successfully unmounted /mnt/state/root-overlay/etc/fake-hwclock.
Feb 04 13:31:56 ff4ffca umount[3481]: umount: /etc/fake-hwclock: not mounted.
```
Cases have also been frequently observed where a filesystem unmount is triggered before a service that still depends on it has been halted. The following log shows `etc-fake\x2dhwclock.mount` unmounting `/etc/fake-hwclock` before `fake-hwclock` has had a chance to save the system time at shutdown (the `bind-etc-fake-hwclock.service` then also tries to unmount the same filesystem):
```
Feb 04 13:51:14 ff4ffca systemd[1]: etc-fake\x2dhwclock.mount: About to execute: /bin/umount /etc/fake-hwclock -c
Feb 04 13:51:14 ff4ffca systemd[1]: etc-fake\x2dhwclock.mount: Forked /bin/umount as 2724
Feb 04 13:51:14 ff4ffca systemd[1]: etc-fake\x2dhwclock.mount: Changed mounted -> unmounting
Feb 04 13:51:14 ff4ffca systemd[1]: Unmounting /etc/fake-hwclock...
Feb 04 13:51:14 ff4ffca systemd[2724]: etc-fake\x2dhwclock.mount: Executing: /bin/umount /etc/fake-hwclock -c
...
Feb 04 13:51:14 ff4ffca fake-hwclock[2723]: [fake-hwclock][INFO] Saving system time to /etc/fake-hwclock/fake-hwclock.data.
Feb 04 13:51:14 ff4ffca fake-hwclock[2733]: /sbin/fake-hwclock: line 52: /etc/fake-hwclock/fake-hwclock.data: Read-only file system
...
Feb 04 13:51:14 ff4ffca systemd[1]: bind-etc-fake-hwclock.service: About to execute: /usr/bin/bindmount --target /etc/fake-hwclock --bind-root /mnt/state/root-overlay --command unmo>
Feb 04 13:51:14 ff4ffca systemd[1]: bind-etc-fake-hwclock.service: Forked /usr/bin/bindmount as 2748
Feb 04 13:51:14 ff4ffca systemd[1]: bind-etc-fake-hwclock.service: Changed exited -> stop
Feb 04 13:51:14 ff4ffca systemd[1]: Stopping Bind mount for /etc/fake-hwclock...
Feb 04 13:51:14 ff4ffca systemd[2748]: bind-etc-fake-hwclock.service: Executing: /usr/bin/bindmount --target /etc/fake-hwclock --bind-root /mnt/state/root-overlay --command unmount
Feb 04 13:51:14 ff4ffca bindmount[2748]: INFO: Unmounting /mnt/state/root-overlay/etc/fake-hwclock ...
Feb 04 13:51:14 ff4ffca bindmount[2748]: INFO: bind mountpont is already unmounted.
```

markcorbinuk, 2021-02-11:
A PR has been generated to replace the `bind-etc-fake-hwclock` service with an `etc-fake\x2dhwclock` mount unit to solve this problem for the `/etc/fake-hwclock` bind mount  - #2114

alexgg, 2021-03-04:
@markcorbinuk can we close this?

markcorbinuk, 2021-03-04:
@alexgg  Although the `/etc/fake-hwclock` bind mount issue was solved, there are still a number of others that could be looked at (from my initial log post):
- umount: /var/volatile/lib/docker: not mounted.
- umount: /var/volatile/log/journal: target is busy.
- umount: /var/lib: not mounted.
- resin-partition-mounter[2495]: INFO: /mnt/data already unmounted.
- umount: /var/volatile: target is busy.

alexgg, 2022-04-29:
This should no longer be a problem - reopen if needed.

closed 2022-04-29 by alexgg