balena-os/meta-balena #2109: docker command is incorrectly named balena

jimmyw, 2021-02-02:
On the devices, the docker command is renamed to balena for an unknown reason. To make the issue more confusing, balena-cli is also having a command called balena that you are using on the host.  So depending on which terminal window you are working on, these commands will give you different results.

For new users, this is very confusing. Please make this right!

Big Thanks!

floion, 2021-02-12:
Hi, this is because the docker daemon you saw renamed as balena in the OS is not just a rename. Please see https://github.com/balena-os/balena-engine#balenaengine-moby-based-container-engine-for-iot

jimmyw, 2021-02-15:
Ok, I did not know that. Still, I mix it up with the balena command on a host. The naming could have been docker-light or something similar. Please feel free to close this issue.

closed 2021-02-18 by klutchell