balena-os/meta-balena #2156: Add data partition fsck to the initrd

mtoman, 2021-04-14:
The data partition's fsck has historically been deferred into a systemd service (`resin-filesystem-expander`) to speed up boot. This is now too late as mobynit will mount the partition before the service is started and the check can not be performed on a moutned partition. The expander has been moved to the initrd for the very same reason, fsck should follow as there is effectively no fsck on the data partition at this moment. This also means the `resin-filesystem-expander` service can be renamed and cleaned up as it still performs some disaster recovery but not the actual expanding.

vipulgupta2048, 2021-07-29:
Just commenting: We have received another failure report for the EXT failures from the JF thread above ^ (4th one in July)

jsheetzati, 2021-09-19:
Hi, we are also looking at SD card failures and are hoping to run fsck on all partitions to see if it corrects the issue. Does this change help us here too? We are using rpi3s.

alexgg, 2021-11-29:
Fixed in v2.87.26

closed 2021-11-29 by alexgg