balena-os/meta-balena #2172: Boot performance has regressed compared to previous versions

jakogut, 2021-04-28:
User reports that from 2.48.0+rev2 to 2.58.6+rev2 the boot performance slowed down, and their container went from starting approximately ten seconds after power on to approximately a minute.

Systemd-udev-settle is known to slow down the boot process, and is deprecated. Bumping plymouth will remove this from several service depends, and allow us to remove it completely.

Several Go binaries are also quite large, and this is at least somewhat of a contributing factor to startup speed. Bumping Go will reduce the size of compiled Go binaries, improving boot speed.

jakogut, 2021-04-29:
I've written a test for this:
```
FROM alpine
CMD /bin/sh -c "echo \"Time from power on to container running: $(awk '{print $1}' /proc/uptime)s\" | tee /dev/kmsg && while true; do sleep 1; done"
```

xginn8, 2021-05-04:
related: https://github.com/balena-os/meta-balena/issues/946

alexgg, 2021-07-15:
This turned out to be Jetson specific - more details on https://jel.ly.fish/75c7a271-7cf1-47e3-b65f-95e882ea2336

closed 2021-07-15 by alexgg

pdcastro, 2021-07-15:
@alexgg, @acostach, @jakogut, so this issue was closed because it was Jetson specific. Are there other GitHub issues or pull requests where the fix was implemented, that could be linked here for reference? Is it perhaps one or both of these PRs? -
* https://github.com/balena-os/balena-jetson/pull/184
* https://github.com/balena-os/balena-jetson/pull/188

I ask because the linked JF support threads reopened when this issue was closed, and I'm trying to figure out what to tell the users. @acostach mentioned the following in one of the threads:

> Hi, we have released a new OS image at version v2.80.5+rev1, can you please test with it and let us know how it works for you?

3 of the 4 linked JF threads are from user accounts of the same customer, but there's another customer who hasn't been notified yet.

acostach, 2021-07-16:
Hi @pdcastro , the v2.80.5+rev1 OS image released for https://jel.ly.fish/75c7a271-7cf1-47e3-b65f-95e882ea2336 is specific to a private jetson device type, which is not part of the balena-jetson repository. I've linked the PR for the private DT in the JF ticket. If the other customer that hasn't been notified is https://jel.ly.fish/11f76f61-acbe-43aa-9c41-defdaa4793fb, then the custom private OS release v2.80.5+rev1 and the BSP related changes in it don't apply.

lorenzave, 2021-07-16:
Hi guys, we are testing v2.80.5 right now. We will report back and see if there are any improvements or issues.
Thanks for the fix :D

lorenzave, 2021-07-19:
The starting of the container improved to nearly half, but not anywhere close to what we had in 2.48 :( 

v2.48: 5/6 seconds for the containers to start
v2.58: 55/60 seconds
v2.80: 34/37 seconds

alexgg, 2021-07-19:
thanks for testing @lorenzave we have opened https://github.com/balena-os/balena-dt-cloudconnector/issues/184 and will be looking into this.