balena-os/meta-balena #2179: update-resin-supervisor exits without fully updating device

20k-ultra, 2021-05-06:
The script for updating the Supervisor will exit if the target version is already on the device. This is not a fair assumption that if the image is downloaded then that Supervisor will be running because something may go wrong after downloading and setting the new version in /etc/resin-supervisor/supervisor.conf.

See how this code exits without checking the if the supervisor.conf is updated as well:

https://github.com/balena-os/meta-balena/blob/b694f0c67f401faee67b51e003641e6b39f4c1a2/meta-balena-common/recipes-containers/resin-supervisor/resin-supervisor/update-resin-supervisor#L143-L150

It's not clear how the script could break in between the steps but a device was seen to have stopped the running Supervisor version, download the new target version, then not update the supervisor.conf file***

***We don't have proof that supervisor.conf was not updated but the logs from update-resin-supervisor show that it is exiting early

```
root@de089a6:~# journalctl -u update-resin-supervisor
-- Logs begin at Thu 2020-08-06 07:25:58 UTC, end at Wed 2021-05-05 17:56:36 UTC. --
May 04 19:16:37 de089a6 sh[3100]: Getting image name and tag...
May 04 19:16:38 de089a6 sh[3100]: Supervisor configuration found from API.
May 04 19:16:38 de089a6 sh[3100]: Getting image id...
May 04 19:16:38 de089a6 sh[3100]: Supervisor registry2.balena-cloud.com/v2/e71173e327e5deb6e7a1d04a859fb491:latest already downloaded.
```

20k-ultra, 2021-05-06:
I'm going to be creating a PR shortly to make the script check what info supervisor.conf has and to update if necessary

20k-ultra, 2021-05-12:
In the case of https://jel.ly.fish/1a010154-2843-4ff2-89d7-d0f94e92b41e the device's supervisor.conf file was actually empty! If we ensure that the version in the configuration file is correct then that device would have be fine.

alexgg, 2021-06-14:
One way we have seen an empty `supervisor.conf` file is when the state partition is full and the `update-resin-supervisor` script tries to update `supervisor.conf`. As the partition is full, the write leaves an empty file.
The state partition contains the rootfs overlay, and on previous versions of BalenaOS some services like NM would leave dangling files on it that with time would completely fill the state partition. Recent versions of BalenaOS do not suffer from this problem.

alexgg, 2021-06-14:
Fixed with https://github.com/balena-os/meta-balena/pull/2211

closed 2021-06-14 by alexgg