balena-os/meta-balena #2194: Allow the dnsServers entry of config.json to override DHCP results

pdcastro, 2021-05-20:
The `dnsServers` entry of `config.json` allows `8.8.8.8` to be excluded, as documented:
https://www.balena.io/docs/reference/OS/configuration/#dnsservers

But it does not allow excluding/overriding results obtained by the device via DHCP. A common need for this is when the user wants to use a locally configured DNS server, in a lab setting for testing and debugging purposes, especially but not limited to the use of [openBalena](https://www.balena.io/open/docs/getting-started/).

Example:

`config.json`
```
  ...
  "dnsServers": "192.168.1.200",
```
In the example above, the **current behavior" is that the device will use DNS servers 192.168.1.1 **and also** 192.168.1.200 (in no configurable order), assuming that 192.168.1.1 was provided to the device via DHCP.

The **desired behavior** would be for the device to use DNS server "192.168.1.200" **only,** ignoring / discarding any DNS servers provided to the device via DHCP.

I think this could be considered a breaking change, so some thought is need as to how to avoid getting devices offline as a result of HUP.

pdcastro, 2021-05-20:
FD discussion: https://www.flowdock.com/app/rulemotion/r-resinos/threads/REGBg4LaGTdax-9beRMc8OZJl8n

majorz, 2022-12-02:
For ignoring the DNS servers passed through DHCP `ipv4.ignore-auto-dns=true` and `ipv6.ignore-auto-dns=true` can be specified inside the NetworkManager connection profile.