balena-os/meta-balena #2201: SV won't start on 2.78.0+rev2 after HUP

acostach, 2021-05-26:
Steps to reproduce:
1) Provision a Pi4 to 2.75.0+rev1
2) Perform HUP either with hostapp-update script or through dashboard in staging to version 2.78.0+rev2
3) After booting in the new OS, check if SV is running and check journal logs:

> May 26 11:40:55 ca5e839 systemd[1]: balena-supervisor.service: Failed to load environment files: No such file or directory
May 26 11:40:55 ca5e839 systemd[1]: balena-supervisor.service: Failed to run 'start-pre' task: No such file or directory
May 26 11:40:55 ca5e839 systemd[1]: balena-supervisor.service: Failed with result 'resources'.
May 26 11:40:55 ca5e839 systemd[1]: Failed to start Balena supervisor.
May 26 11:41:49 ca5e839 systemd[1]: balena-supervisor.service: Failed to load environment files: No such file or directory
May 26 11:41:49 ca5e839 systemd[1]: balena-supervisor.service: Failed to run 'start-pre' task: No such file or directory
May 26 11:41:49 ca5e839 systemd[1]: balena-supervisor.service: Failed with result 'resources'.
May 26 11:41:49 ca5e839 systemd[1]: Failed to start Balena supervisor.
May 26 11:41:59 ca5e839 systemd[1]: balena-supervisor.service: Failed to load environment files: No such file or directory
May 26 11:41:59 ca5e839 systemd[1]: balena-supervisor.service: Failed to run 'start-pre' task: No such file or directory
May 26 11:41:59 ca5e839 systemd[1]: balena-supervisor.service: Failed with result 'resources'.
May 26 11:41:59 ca5e839 systemd[1]: Failed to start Balena supervisor.
May 26 11:42:09 ca5e839 systemd[1]: balena-supervisor.service: Failed to load environment files: No such file or directory
May 26 11:42:09 ca5e839 systemd[1]: balena-supervisor.service: Failed to run 'start-pre' task: No such file or directory
May 26 11:42:09 ca5e839 systemd[1]: balena-supervisor.service: Failed with result 'resources'.
May 26 11:42:09 ca5e839 systemd[1]: Failed to start Balena supervisor.

Reproducible on other device types too.

acostach, 2021-05-26:
@klutchell is this a known issue, are there some other pending changes related to sv renaming in meta-balena?

klutchell, 2021-05-26:
@acostach There are no outstanding changes regarding the SV rename, and this is the first time I've seen the errors above. Some HUP testing was performed before the merge.

klutchell, 2021-05-28:
It appears that the supervisor image is not updated during hostapp-update so after upgrading from `2.75.0+rev1` to `2.78.0+rev2` the supervisor is still attempting to start `v12.5.10` when it must be at least `v12.6.8` to support the new environment file path and service name.

I guess we have a condition where the old supervisor needs to start before downloading the new supervisor? I thought this would have been done on first boot after a hostapp-update?

klutchell, 2021-05-28:
During testing I didn't find this was needed, but looking at it again I think we need to support both supervisor.conf locations in resin-state-reset.

https://github.com/balena-os/meta-balena/blob/3fd49a2d6882abf938be0f7aadedaddb085fa084/meta-balena-common/recipes-support/resin-state-reset/resin-state-reset/resin-state-reset#L14-L17

I will try some tests and see what happens.

closed 2021-06-03 by bulldozer-balena[bot]

acostach, 2021-06-09:
@klutchell looks there are still some related issues with the sv renaming, the supervisor doesn't get updated during hup from dashboard and the old supervisor runs in the new OS. The latest tag appears to be pulled but the supervisor.conf still has the old version. Happens on every HUP, tested on a Pi4.