balena-os/meta-balena #2244: Potential race condition when setting hostname at boot

markcorbinuk, 2021-07-08:
There is a potential race condition when setting the device hostname at boot. The `resin-hostname` service runs asynchronously to the `NetworkManager` service with both services attempting to set the transient hostname. This is potentially only an issue at first boot when `/etc/hostname` is empty or when re-booting after having modified/added a hostname field in `config.json`.

markcorbinuk, 2021-07-20:
TL;DR The hostname race condition can lead to a mismatch between the transient and static hostname settings as per the scenarios described below.

The following conditions have been verified on a RaspberryPi 3 B+ running under balenaOS 2.80.8+rev4 (meta-balena 2.82.9).

**1. When the hostname is updated via changes to `config.json` (either manually or via the supervisor API)**

The sequence at boot is as follows:

- NetworkManager reads the 'oldhostname' from /etc/hostname.
- The `balena-hostname` service then sets both the transient and static (/etc/hostname) hostname to 'newhostname' read from config.json.
- NetworkManager sets the hostname using the value read previously, i.e. 'oldhostname'.

Journal logging sequence is as follows:
```
Jul 20 13:34:58 localhost   (some log messages)
Jul 20 13:34:58 locahost    NetworkManager starts and reads 'oldhostname' from /etc/hostname
Jul 20 13:34:58 localhost   The balena-hostname service runs and sets 'newhostname' (transient and static) from config.json
Jul 20 13:34:58 newhostname (some log messages)
Jul 20 13:34:58 newhostname Avahi service starts broadcasting 'newhostname.local'
Jul 20 13:34:58 newhostname (some log messages)
Jul 20 13:34:58 newhostname NetworkManager sets 'oldhostname'
Jul 20 13:34:58 oldhostname (some log messages)
Jul 20 13:34:58 oldhostname (some log messages)
```
This issue affects the journal logs and the console prompt. Broadcast via mDNS appears to work because avahi is started immediately after the `balena-hostname` service. However, the broadcast name would be incorrect if avahi was restarted at any point.

Console prompt shows `root@oldhostname:~#`
Journal messages show `oldhostname`
Avahi is broadcasting `newhostname.local`

**2. When a newly provisioned device is started (i.e. a device with an empty /etc/hostname)**

The sequence at boot is as follows:

- NetworkManager attempts to read  a hostname from /etc/hostname and fails.
- The `balena-hostname` service then sets both the transient and static (/etc/hostname) hostname to 'newhostname' read from config.json.
- NetworkManager sets the hostname based on a 'best-guest' approach as it previously failed to find a valid hostname. It seems that this is typically done using a reverse DNS lookup.

Journal logging sequence is as follows:
```
Jul 20 13:34:58 localhost   (some log messages)
Jul 20 13:34:58 locahost    NetworkManager starts and fails to read a hostname from /etc/hostname
Jul 20 13:34:58 localhost   The balena-hostname service runs and sets 'newhostname' (transient and /etc/hostname) from config.json
Jul 20 13:34:58 newhostname (some log messages)
Jul 20 13:34:58 newhostname Avahi service starts broadcasting 'newhostname.local'
Jul 20 13:34:58 newhostname (some log messages)
Jul 20 13:34:58 newhostname NetworkManager sets 'newhostname' based on reverse DNS lookup
Jul 20 13:34:58 newhostname (some log messages)
Jul 20 13:34:58 newhostname (some log messages)
```
Although NetworkManager always appears to set the correct 'newhostname' via reverse DNS when booting with a network connection, it is not known under what conditions this may fail. When booting without a network connection NetworkManager falls back to setting an incorrect hostname of `localhost.localdomain`.

klutchell, 2021-07-20:
Not sure if this helps with the fix at all, but there is a new way to have services restart when config.json changes.

PR: https://github.com/balena-os/meta-balena/pull/2250

Just adding `PartOf=config-json.target` will trigger the service to restart when config.json changes. There's obviously other race conditions going on here, but I bet this method will come in handy at some point.

markcorbinuk, 2021-07-20:
@klutchell Yes, I spotted that and the work you have done with dnsmasq. It will definitely help.

Once I've straightened out the dependencies it would be good to auto-set the hostname and restart the avahi service when config.json is changed