balena-os/meta-balena #2257: Support Realtek 8188EUS chipset

majorz, 2021-07-16:
Reportedly the 8188EUS chipset is working on other OSes like Raspberry Pi OS, but not on balenaOS.

majorz, 2021-07-19:
I happened to have two dongles with that chipset. One is the official Raspberry Pi USB WiFi Dongle (now discontinued), the other one is a TP-Link TL-WN725N dongle.

This is how the RPi one shows on Arch linux:
```
T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  7 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=8179 Rev=00.00
S:  Manufacturer=Realtek
S:  Product=802.11n NIC
S:  SerialNumber=00E04C0001
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=r8188eu
```

majorz, 2021-07-20:
```
[41536.616075] usb 1-1: new high-speed USB device number 12 using xhci_hcd
[41536.759786] usb 1-1: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[41536.759808] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[41536.759820] usb 1-1: Product: 802.11n NIC
[41536.759829] usb 1-1: Manufacturer: Realtek
[41536.759838] usb 1-1: SerialNumber: 00E04C0001
[41536.807324] lib80211: common routines for IEEE802.11 drivers
[41536.807333] lib80211_crypt: registered algorithm 'NULL'
[41536.856828] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[41536.861550] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[41536.883598] usbcore: registered new interface driver r8188eu
[41537.269783] MAC Address = c0:25:e9:1d:4b:39
[41537.274825] R8188EU: indicate disassoc
```

majorz, 2021-07-21:
I did not find much difference between Arch, Raspberry Pi OS and balena. In all cases the same driver gets loaded, the only dissimilarity is between kernel versions. Theoretically the newer the kernel, the more stable the driver should be.

floion, 2022-01-12:
These are the PRs for getting these dongles working with balenaOS:
https://github.com/balena-os/meta-balena/pull/2361
https://github.com/balena-os/meta-balena/pull/2362