balena-os/meta-balena #2304: Check if target Supervisor version in recipe is available

20k-ultra, 2021-09-01:
To update the default Supervisor version we will set `SUPERVISOR_TAG` to the target version: https://github.com/balena-os/meta-balena/blob/1b9f4915ff3e1ac874e6493dc21af444bf97bc14/meta-balena-common/recipes-containers/balena-supervisor/balena-supervisor.inc#L20

This PR can be merged with tests passing but this would break HUP if that version does not exist from the cloud environment which HUP pulls the images from. This previously wasn't an issue because the images were pulled from docker hub and Supervisor images were automatically built then pushed there but now we pull the images from balenaCloud and the Supervisor repo does not currently have automated deployments yet.

We should add some test coverage to check if this Supervisor version is available in the target cloud environment that HUP will try to pull it from.

[this](https://github.com/balena-os/balenahup/blob/2a4c15ce3cf350a76d6d2a1765a7e8df168035c4/upgrade-2.x.sh#L223-L224) is the line from HUP that fails if the version does not exist. We might be able to simply re-use this somewhere.

klutchell, 2021-09-02:
@20k-ultra Are we still pushing supervisor images to dockerhub on every merge to master?

We must be or else the yocto build would have failed here.

https://github.com/balena-os/meta-balena/blob/47d4f60537622b3bccf63b87f809d114e0085097/meta-balena-common/recipes-containers/docker-disk/docker-disk.bb#L17-L19

https://github.com/balena-os/meta-balena/blob/47d4f60537622b3bccf63b87f809d114e0085097/meta-balena-common/recipes-containers/docker-disk/files/entry.sh#L49-L50

So the main issue I see is that the "preload" of the supervisor during the yocto build still uses DockerHub. This will change when we move to supervisor as an app. 

If we want to test this anyway, we could add an automated test to run `update-balena-supervisor` and check for errors. We don't expect it to update since it should be running the target release, but we also don't want an error that the image doesn't exist in our registry.

20k-ultra, 2021-11-23:
we can close this since the Suepervisor has automated deployments that make versions always available in staging and prod.

closed 2021-11-23 by 20k-ultra