balena-os/meta-balena #2319: Disabling user namespacing by default makes chromium based application to fail

alexgg, 2021-09-15:
On commit https://github.com/balena-os/meta-balena/commit/04d9561760b449024a5b411bd366ded2351cd5cd we forced user namespacing support. However, as it might expose host vulnerabilities, we disabled it by default in:
https://github.com/balena-os/meta-balena/commit/31c3ae8ad5c7ad45e450349b6972524da120e96c

However, some BSPs were alreay enabling the config beforehand, so by default to disable we are making some applications, specifically chromium, not work unless the applications are modified to change this setting. This can only be modified from privileged applications.

rahul-thakoor, 2021-09-15:
related to : https://github.com/balenablocks/browser/issues/77

alexgg, 2021-12-27:
This has been addressed in v2.85.10

closed 2021-12-27 by alexgg