balena-os/meta-balena #2326: Engine healthcheck unable to complete on systems with very high CPU load / IO throughput

alexgg, 2021-09-20:
On systems that are heavily loaded, both in terms of CPU load and IO throughput, the engine is not even able to perform a healthcheck in the 6 mins timeout. When the engine is killed, it is also not able to close down on time so it is forcefully aborted which in turns might corrupt the engine storage and leave the engine not able to start at all.

alexgg, 2021-09-20:
Fixes https://github.com/balena-os/meta-balena/issues/1384

splitice, 2021-09-23:
I can confirm I've seen this too. It's not uncommon during container updates on systems that are already heavily loaded.

alexgg, 2021-09-23:
Relates to https://github.com/balena-os/meta-balena/issues/2327

jrc, 2021-11-29:
Balena support provided the following workaround:

> Hello John,
> 
> RasPi Zero's are definitely the devices that are the most susceptible ones to this issue.
> 
> I assume you application puts the device under heavy load (for a Zero at least) when this happens, is that correct?
> 
> Until we release a fix for this issue, there is a workaround, which you can implement. But beware that you **must undo this change before the next OS update** otherwise it will be rejected.
> The steps are as follows:
> 
> * Remount the root filesystem as read-write: `mount -o remount,rw /`
> * Run systemctl edit --full balena
> * Add a WatchdogSec entry to the Service section of the unit file. You may need to create the section if it is not present. Like this:
> 
>       [Service]
>       WatchdogSec=720
> * Remount the root filesystem as read-only: `mount -o remount,ro /`
>
> That changes the watchdog to allow up two 12min to respond. If that's not enough, you can increase that number, but that would be unexpected, and we would like to know about it if you deem that neccessary.

klutchell, 2022-01-19:
Relates to https://github.com/balena-os/meta-balena/issues/1397

alexgg, 2022-08-11:
@lmbarros we can probably close this once #1397 is merged

lmbarros, 2022-08-11:
> we can probably close this once #1397 is merged

Yes, thanks. Just added a "Resolves" link.

closed 2022-08-17 by bulldozer-balena[bot]