balena-os/meta-balena #2327: balenaEngine cannot handle SIGABRT sent by default by the watchdog

lmbarros, 2021-09-22:
By default, systemd's watchdog will send a `SIGABRT` to kill processes, but the Go runtime uses this signal to trigger a panic, and therefore the usual recommendations is to not handle it on Go programs. As a result, whenever the watchdog kills balenaEngine, it doesn't have the chance to do any cleanup.

This is a more general issue, but is suspected to cause problems in one concrete case: upon being `SIGABRT`ed, balenaEngine apparently leaves overlayfs mounts mounted. When the Engine is restarted, it mounts again the same points, which results on some warnings (like `overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.`), and supposedly can cause further problems down the road.

_Maybe_ this is just a matter of adding `WatchdogSignal=SIGTERM` to [balena.service](https://github.com/balena-os/meta-balena/blob/master/meta-balena-common/recipes-containers/balena/balena/balena.service), but we really should look deeper into this because:

* We need to check if balenaEngine properly handles `SIGTERM` on the first place, doing the required clean ups.
* We might want to send a `SIGKILL` if the `SIGTERM` doesn't kill balenaEngine after some time. (This might need some additional settings on the systemd unit).
* And if we really send a `SIGKILL` sometimes, we'd be back to the current situation in which balenaEngine would not clean up stuff. Maybe we need a better plan overall.

lmbarros, 2021-09-22:
I tried to reproduce this on a Raspberry Pi Zero, but could not get exactly the same errors we have seen before. Here's a summary.

What I did: I had a Pi Zero running some CPU-intensive tasks. I `pushed` and update that would create more of these CPU-intensive containers. Just after my new images were downloaded (that is, before the new containers were created), I manually `kill -ABRT`ed the `balenad` process and watched what happened:

* balenaEngine panicked. This is expected, as this is the default behavior of Go programs getting a `SIGABRT`.
* Very soon we get a number of `connect ECONNREFUSED /var/run/balena-engine.sock` errors on the logs, coming from the Supervisor container. This looks expected to me: the Supervisor, I suppose, is trying to talk with a broken Unix socket (the process listening on the other side is no more).
* balenaEngine will be restarted. Upon restart, we see some systemd warnings telling about left-over processes. Initially, these left-over processes correspond to my user-containers that were previously running -- but more on this soon!
* balenaEngine slowly started to run my new containers (as per the update I had triggered). This put additional load into the system, so eventually:
```
	Sep 22 19:46:25 87dbffb systemd[1]: balena.service: start operation timed out. Terminating.
	Sep 22 19:46:43 87dbffb systemd[1]: balena.service: Failed with result 'timeout'.
	Sep 22 19:46:43 87dbffb systemd[1]: Failed to start Balena Application Container Engine.
```
* I am not sure how systemd does this check and the service restart. I certainly did not see any other panics on the logs, so it was not a `SIGABRT`. Maybe for initialization timeouts systemd sends a `SIGSTOP` instead?
* So, we entered a crash loop. balenaEngine would keep starting new containers (those I asked for!), but would end up being killed again because "start operation timed out."
* During this crash loop, the list of left-over processes on the logs was getting longer at every iteration. Some of this is expected: as before, these are the containers I really wanted to have running. balenaEngine seems to be doing the right thing with them (it keeps them running, and even detects them to be in containers that it is supposed to manage).
* But some of these left-over processes are just the failed health checks -- those that, I believe, systemd was running to check if the initialization completed. Those processes remain running forever.
* Eventually, the device stabilized. All user containers were created, and balenaEngine reported as healthy to systemd. This ended the crash loop. Except for those rogue healthcheck processes I mentioned just above, the system seemed perfectly healthy to me.
* At no point I saw the overlayfs warnings. I was expecting to see those.

I tried this a couple more times with the same results. I also tried a `SIGSTOP` once, and the only difference I noticed was that we got no Go panic (as expected: a `SIGSTOP` will not trigger a panic).

toochevere, 2021-09-29:
@lmbarros If I understand correctly, the difference between your scenario that you were able to create versus what is happening with the customer is that your crashloop had periods of lengthening stability which eventually allowed the system to fully stabilize, right?  If so, could it be that this is really pretty close to the issue and for whatever gremlins may be, wasn't exactly the same?  Do we still want to fix the case of crashlooping that you found?

lmbarros, 2021-09-29:
@toochevere We have seen a few different cases involving crashloops with slightly different symptoms. In the test above I was specifically testing an hypothesis we had to explain the `overlayfs` errors we have seen sometimes. We are still investigating, I can't tell now if they are related or not.

But sure, you right in that the steps I followed above are good enough to reproduce a real issue we must fix.

closed 2022-08-17 by bulldozer-balena[bot]