balena-os/meta-balena #2331: Test "TC139 - SSH login" fails on rpi4 version 2.85.2+rev3

floion, 2021-09-24:
Setting `developmentMode` to `true` in config.json and then adding a valid or invalid public ssh key to `os.sshKeys` in config.json still allows `ssh root@<IP_ADDR>`

Setting `developmentMode` to `true` in config.json and then adding a valid public ssh key to `os.sshKeys` in config.json does not allow `ssh <USER>@<IP_ADDR>`

alexgg, 2021-09-24:
@floion  as discussed "v2.85.2" needs a reboot to apply "os.sshKeys" configuration changes, this is addressed in "v2.85.3"

If you are doing:

```
ssh -p 22222 root@<IP_ADDR>
```
But you configured the `os.sshKeys` with your user's (non-root) keys, you will need to specify which key to use as in:
```
ssh -p 22222 -i ~/.ssh/id_rsa.pub <USER>@<IP_ADDR>
```

alexgg, 2021-11-23:
@floion I believe this was sorted by the comment above, please re-open otherwise.

closed 2021-11-23 by alexgg