balena-os/meta-balena #2349: resindataexpander: retry filesystem resize if only the partition has been resized

mtoman, 2021-10-20:
At this moment `resindataexpander` will only execute if the actual partition on the block device needs to be resized. In that case it
* resizes the partition
* resizes the filesystem

If the partition is resized successfully but the filesystem resize fails for any reason, it would never retry and the logs from resize2fs are gone after first boot.

These two should be independent:
* Step 1: check whether the partition needs expanding and do it
* Step 2: check whether the filesystem needs expanding and do it

mtoman, 2021-11-11:
Consider extending this for LUKS resize as well

alexgg, 2022-04-29:
This should be fixed by https://github.com/balena-os/meta-balena/pull/2317

closed 2022-04-29 by alexgg