balena-os/meta-balena #2423: Engine healthchecks cause additional wear to storage media

klutchell, 2021-11-26:
Running `iostat` and `inotifywait` on the data partition shows regular docker layer create/delete and network database writes on an otherwise idle engine.

The engine healthcheck every 6 minutes would definitely result in some of these writes, if not all of them.

klutchell, 2021-12-03:
Confirmed with the following steps that with user apps and supervisor disabled, the only writes are from the engine healthcheck.

With the healthcheck extended to 1h, there were no writes at all.

```bash
# Stop all user services via the dashboard stop icons

# Stop the supervisor
systemctl stop update-balena-supervisor.timer
systemctl stop balena-supervisor.service

# Extend engine healthcheck to 1h intervals
mkdir -p /run/systemd/system/balena.service.d
cat << EOF > /run/systemd/system/balena.service.d/override.conf
[Service]
WatchdogSec=3600
EOF

# Reload the daemon
systemctl daemon-reload

# Check the change has been applied
systemctl show balena | grep WatchdogUSec

# Restart the service
systemctl restart balena

# Run inotifywatch to gather filesystem access statistics for 10m
balena run --rm -it -v /mnt/data:/mnt/data:ro alpine sh -c \
  'apk add --no-cache inotify-tools && /usr/bin/inotifywatch -v -t 600 -r /mnt/data'
  ```

alexgg, 2022-08-11:
@lmbarros maybe you can still add a "Fixes #2423" comment you your PR above.

lmbarros, 2022-08-11:
It's there already (using "Resolves")!

closed 2022-08-17 by bulldozer-balena[bot]