balena-os/meta-balena #2438: RPI3: leviathan hup tests fail starting with v2.86.18, blocking all PRs in balena-raspberrypi

acostach, 2021-12-07:
The HUP test fails on the Pi3 when pushing the update image to the registry.

internal discussion: https://www.flowdock.com/app/rulemotion/device-testing/threads/vIOuQkPUTZFcZbWDl8WOaUnkQnI
Reproduced locally by Vipul.

PRs which always fail:
https://github.com/balena-os/balena-raspberrypi/pull/740
https://github.com/balena-os/balena-raspberrypi/pull/735 (superseded, updated from v2.86.16 to v2.86.18)

These PRs only update the meta-balena layer in the device repsository.

acostach, 2021-12-07:
A reboot of the testbot done by Vipul solved the problem for the meta-raspberrypi v2.88.4 PR.

klutchell, 2021-12-07:
Closing for now as we plan on moving the registry off the DUT going forward.

closed 2021-12-07 by klutchell