balena-os/meta-balena #2439: Investigate why the device fails to reboot after RAID1 provisioning

mtoman, 2021-12-08:
There is no supported procedure for provisioning balenaOS on a RAID1 MD array but the OS supports booting off it. Assembling the array manually and copying the image to it kind of works but the device does not come up after reboot on certain hardware. This needs to be investigated while designing an official provisioning procedure.

alexgg, 2022-08-11:
@jakogut please link to the RAID improvement

jakogut, 2022-09-13:
I think this one is closed by https://github.com/balena-os/meta-balena/pull/2761

The above PR documents how to setup a RAID 1 array that is bootable by balenaOS. I suspect the issue with the array not being bootable on some devices is an MBR/GPT compatibility issue with different metadata versions.

closed 2022-09-13 by jakogut