balena-os/meta-balena #2459: Update Supervisor.conf atomically

20k-ultra, 2021-12-23:
The way we update the Supervisor config file is not atomic https://github.com/balena-os/meta-balena/blob/ba552417d7be602afbb636b3b64781e6dae8aece/meta-balena-common/recipes-containers/balena-supervisor/balena-supervisor/update-balena-supervisor#L147

20k-ultra, 2021-12-23:
`sed -i` is actually doing exactly what I was working on adding (writing to a tmp file then moving). Therefore, the `-i` flag makes the update atomic. We can improve this though by syncing the cached writes to disk.

jakogut, 2022-01-19:
@20k-ultra Hypothetically, if `sed -i` is writing to a temp file and renaming it, the old file should be in place until the rename is flushed to disk. We can explicitly flush to disk using `sync`, but there should never be a point where the file is in an inconsistent state from this particular write.

jakogut, 2022-01-19:
Looking at one of the corrupted file examples, this looks more likely to be an unreliable SD card than a partial/interrupted write, as the file is full of a lot of binary junk rather than text.

alexgg, 2022-04-29:
The conclusion is that the update is safe to do in the way it is, so closing this.

closed 2022-04-29 by alexgg