balena-os/meta-balena #2464: `config.json` hostname changes in OS 2.85.2+rev3.dev restart OpenVPN, balenaEngine, and reset SSH connections

cywang117, 2022-01-04:
Is this intended behavior?

Environment:
- RPi4
- OS 2.85.2+rev3, development
- Supervisor 12.10.10

To reproduce, change hostname in `config.json`.

Logs: https://gist.github.com/cywang117/f122c75ddd8e9042b11542547f35f886

alexgg, 2022-01-05:
@cywang117 changes in config.json make services that configure viaconfig.json restart. This avoids having to reboot to apply configuration changes. There is no mechanism currently to discriminate and only restart services that are affected by the configuration change.

Is this causing any unintended behaviour?

maggie0002, 2022-01-05:
When I logged the original question that led to this, I was looking to understand what the default behaviour is to allow me to anticipate what sort of user interaction can be expected. The behaviour you described is actually what I was hoping was the case. 

The only X factor here, is whether the hostname inside a container mounted to `host` is changed too (i.e. typing `hostname` at the console inside the container), or does that require a container rebuild (which it seems is also managed by Balena OS and/or the supervisor, but can't seem to nail down the default behaviour for that either)?

cywang117, 2022-01-05:
@maggie0002 When I tested a hostname change via Supervisor API in 2.83, where `config.json` wasn't watched and services didn't restart on config changes, the containers maintained their old hostnames. To see a new hostname in container, the container has to be rebuilt, which is handled by the engine service restart in OS 2.85.

klutchell, 2022-03-25:
This has been resolved by https://github.com/balena-os/meta-balena/pull/2495

closed 2022-04-29 by alexgg