balena-os/meta-balena #2471: Enable serial in RPi devices before flashing

rcooke-warwick, 2022-01-12:
Since rpi os version 2.88.5+rev1, we must now set `enable_uart=1` in the `config.txt` of the image for it to output serial. We want to do this before flashing so we can capture serial on first boot, in case the device has a problem booting.

We should also add a Rpi specific test that checks explicitly if we are capturing serial output

klutchell, 2022-01-12:
We can do this by patching the target state via supervisor, like we do for the device-tree tests.
https://github.com/balena-os/meta-balena/blob/d50cb831fac22b46c16a9b949e2657d0bb199858/tests/suites/os/tests/device-tree/index.js#L77-L104

klutchell, 2022-01-12:
From @vipulgupta2048 , we may be able to use imagefs to write the file before booting.

https://github.com/balena-os/leviathan/blob/c5d8243b93a3789bcac164415ca36bd884269abb/core/lib/components/os/balenaos.js#L114-L122

closed 2022-01-19 by bulldozer-balena[bot]