balena-os/meta-balena #2512: Failed execute balena-image:do_image_hostapp_ext4

sprhawk, 2022-02-16:
Yocto/Poky version: Hardknott

```
| time="2022-02-12T00:33:43.024183309Z" level=info msg="API listen on /var/run/balena-host.sock"
| + hostapp-update -f /input -n
| + kill 7
| + wait 7
| time="2022-02-12T00:33:49.549516715Z" level=info msg="Processing signal 'terminated'"
| time="2022-02-12T00:33:49.549919765Z" level=info msg="stopping event stream following graceful shutdown" error="<nil>" module=libcontainerd namespace=moby
| time="2022-02-12T00:33:49.549938997Z" level=info msg="Daemon shutdown complete"
| time="2022-02-12T00:33:49.549970717Z" level=info msg="stopping healthcheck following graceful shutdown" module=libcontainerd
| time="2022-02-12T00:33:49.550002624Z" level=info msg="stopping event stream following graceful shutdown" error="context canceled" module=libcontainerd namespace=plugins.moby
| + mkfs.ext4 -F -E lazy_itable_init=0,lazy_journal_init=0 -i 8192 -d /mnt/sysroot/inactive /output
| mke2fs 1.43.4 (31-Jan-2017)
| Discarding device blocks: done
| Creating filesystem with 327680 1k blocks and 40960 inodes
| Filesystem UUID: 5e006b2c-bd0b-475d-9cd8-73c7d96d47a3
| Superblock backups stored on blocks:
|       8193, 24577, 40961, 57345, 73729, 204801, 221185
|
| Allocating group tables: done
| Writing inode tables: done
| Creating journal (8192 blocks): done
| Copying files into the device: __populate_fs: Could not allocate block in ext2 filesystem while writing file "mac80211.ko"
| mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system
| Deleted: sha256:f60d82642878c25023cff0edbf686a9ee800f2a0c4aea47ee838e8ea30be6274
| Deleted: sha256:2c772767545356ca03e83c34e61006fcfca908a1e7d299cc2979c1751709bc7c
| Deleted: sha256:acc14bfa0fe6917af21bac4cddd26629a75244190380355983fee03a1141b0aa
| WARNING: exit code 1 from a shell command.
```
[log.zip](https://github.com/balena-os/meta-balena/files/8077600/log.zip)

alexgg, 2022-04-29:
hi @sprhawk is this also failing in Honister? What are the build commands you are using and for what device type?

alexgg, 2022-08-11:
Closing as this does not occur on Honister and Hardknott was never supported.

closed 2022-08-11 by alexgg