balena-os/meta-balena #2533: Missing sftp-server

hgaiser, 2022-03-08:
I was trying to use `sshfs` for manipulating files on a balena machine running `balenaOS 2.83.18+rev1`, but I get an error saying `remote host has disconnected`. Using either `ssh` or `scp` does work.

In `/etc/ssh/sshd_config` on the balena machine I see the following:

```
Subsystem       sftp    /usr/libexec/sftp-server
```

Which, as far as I understand it, defines which binary to execute for running an sftp-server (the protocol used in [`sshfs`](https://linux.die.net/man/1/sshfs): "On the remote computer the SFTP subsystem of SSH is used."). The problem is that `/usr/libexec/sftp-server` does not exist. I can work around this issue by using `scp`, but using `sshfs` is much more convenient in my opinion. I can host a service inside the container that runs the sftp-server, but then I have to manage the keys in that container as well, which is a hassle.

Is there a reason this `sftp-server` is missing? Or is this by accident? It should be installed together with [`openssh`](https://www.openssh.com/), which seems to be how `sshd` is installed.

alexgg, 2022-04-18:
Hi Hans, as you mention the reason not to include `sftp-server` in the hostOS is because nothing in the hostOS requires it. As you say, production devices that need it should add it to the application container, and OS developers that need it can build an image as they see fit.

hgaiser, 2022-04-18:
I see, that's unfortunate because as I stated earlier it means I have to manage the SSH keys in that container too, which is already handled by balena. I think it would be best to include the sftp server in develop builds only, would that be an option? This is mainly for easier development, not for production systems.

alexgg, 2022-05-18:
hi @hgaiser, there are no longer development images, it's now a runtime configurable development mode. We have discussed this internally and development mode is really application development. OS/BSP developers have the knowledge and resources to build custom development images as they see fit.

alexgg, 2022-05-18:
Anyway, right now the rootfs space is tight for some device types, but as soon as space is available (for example once we can move some modules and firmware to extensions) we will consider the addition. See also https://github.com/balena-os/meta-balena/issues/2618

closed 2022-05-18 by alexgg