balena-os/meta-balena #2580: init_config_json not detecting and applying development mode properly

dcaputo-harmoni, 2022-04-13:
It looks like there is a bug in image-balena.bbclass where it is not correctly detecting a development image.  Below is a patch with the proposed change which is working for me.

```
diff --git a/meta-balena-common/classes/image-balena.bbclass b/meta-balena-common/classes/image-balena.bbclass
index 70ff852d..658a9386 100644
--- a/meta-balena-common/classes/image-balena.bbclass
+++ b/meta-balena-common/classes/image-balena.bbclass
@@ -60,9 +60,8 @@ init_config_json() {
    # Set deviceType for supervisor
    echo "$(cat ${1}/config.json | jq -S ".deviceType=$slug")" > ${1}/config.json

-   if ${@bb.utils.contains('DISTRO_FEATURES','osdev-image','true','false',d)}; then
-      echo "$(cat ${1}/config.json | jq -S ".developmentMode=true")" > ${1}/config.json
-   fi
+   # Set developmentMode based on osdev-image
+   echo "$(cat ${1}/config.json | jq -S ".developmentMode=${@bb.utils.contains('DISTRO_FEATURES','osdev-image','true','false',d)}")" > ${1}/config.json
 }

 #
```

alexgg, 2022-04-18:
hi @dcaputo-harmoni thanks for your comment.

To build a development image you can use the `-d` flag to barys. That will set an `OS_DEVELOPMENT` variable in `config.json` that will build images suitable for BSP development(see https://github.com/balena-os/meta-balena#os-development). That will set the `DISTRO_FEATURES` to `osdev-image`.

Images deployed to balenaCloud are automatically configured in development or production mode based on the user requests via the CLI/SDK or API.

closed 2022-04-18 by alexgg

dcaputo-harmoni, 2022-04-18:
@alexgg I am aware of how to build a development image but my comment relates to the specific lines of code identified in the issue. You are using a logical test on the string “true” which doesn’t work - so the revised code uses the string directly instead.

alexgg, 2022-04-22:
hi @dcaputo-harmoni I'm afraid I don't know what you mean. That type of construct is valid and used in multitude other places in Poky, for example:
https://git.yoctoproject.org/poky/tree/meta/classes/rootfs-postcommands.bbclass?h=honister#n122

But also, I added some debug prints and ran both images configured with and without `OS_DEVELOPMENT` and verified the switch is working.

So what do you mean exactly when you say ` logical test on the string “true” which doesn’t work`? Did you add some debug prints and verified this, or are you not seeing the `developmentMode` on your generated `config.json` file?

dcaputo-harmoni, 2022-04-24:
Using debug statements during the build process, `${@bb.utils.contains('DISTRO_FEATURES','osdev-image','true','false',d)}` prints "true", however code inside the `if ${@bb.utils.contains('DISTRO_FEATURES','osdev-image','true','false',d)}; then` block does not execute (verified both using debug statements and confirming that `developmentMode` does not get set to true).

However, when using the variable directly to set `developmentMode` it works, i.e. `echo "$(cat ${1}/config.json | jq -S ".developmentMode=${@bb.utils.contains('DISTRO_FEATURES','osdev-image','true','false',d)}")" > ${1}/config.json`

alexgg, 2022-04-29:
hi @dcaputo-harmoni what Linux distribution and shell are you using?

dcaputo-harmoni, 2022-04-29:
@alexgg I am using the Jenkins build process which utilizes `balena-yocto-scripts/automation)/Dockerfile_yocto-build-env` for the yocto build.