balena-os/meta-balena #2581: NetworkManager occasionally fails to activate AP profile properly

majorz, 2022-04-13:
When NetworkManager creates an access point with Internet connection sharing enabled it adds a number of iptables rules. Such an access point profile activated on boot may partially fail due to collision with balenaEngine. balenaEngine at the same time creates iptables rules as well, but holds the xtables.lock file, so some of the rules NetworkManager creates may fail. The problem looks to be within NetworkManager is it does not use `-w, --wait` option of iptables.

This is yet to be reproduced locally by us. After we reproduce it and confirm that the missing wait of NetworkManager's usage of iptables is the root of the issue, we may file a bug report towards NetworkManager and take it from there.

Current workaround is to create the AP profile from a container or at least reactivate it there.

An example log that illustrates the failure (it was reported that this can happen on arbitrary rules):
```
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.3579] Executing: /usr/sbin/iptables --table filter --insert INPUT --in-interface wlan0 --protocol tcp --destination-port 53 --jump ACCEPT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.3666] Executing: /usr/sbin/iptables --table filter --insert INPUT --in-interface wlan0 --protocol udp --destination-port 53 --jump ACCEPT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.3769] Executing: /usr/sbin/iptables --table filter --insert INPUT --in-interface wlan0 --protocol tcp --destination-port 67 --jump ACCEPT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.3867] Executing: /usr/sbin/iptables --table filter --insert INPUT --in-interface wlan0 --protocol udp --destination-port 67 --jump ACCEPT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.3971] Executing: /usr/sbin/iptables --table filter --insert FORWARD --in-interface wlan0 --jump REJECT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.4172] Executing: /usr/sbin/iptables --table filter --insert FORWARD --out-interface wlan0 --jump REJECT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.4306] Executing: /usr/sbin/iptables --table filter --insert FORWARD --in-interface wlan0 --out-interface wlan0 --jump ACCEPT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.4387] Command returned exit status 4.
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.4389] Executing: /usr/sbin/iptables --table filter --insert FORWARD --source 10.42.0.0/255.255.255.0 --in-interface wlan0 --jump ACCEPT
Apr 05 22:00:12 9e8f192 NetworkManager[1571]: [1649196012.4478] Command returned exit status 4.
```

majorz, 2022-11-21:
Related is https://github.com/balena-os/meta-balena/issues/2872 as we need to update NetworkManager to latest stable before patching it as that part of the code base was refactored.

majorz, 2022-12-19:
I could not reproduce manually, so I created a small application to automate the process: https://github.com/balena-io-experimental/iptables-racing-2581. Going to try this in different ways.

majorz, 2022-12-20:
So far I have not been able to reproduce the racing condition on a few device types with different NetworkManager versions. I am going to try some extra ideas.

louisburton, 2022-12-20:
If able to reproduce this here as has been done by Balena support before, happy for you to use this for testing -
https://dashboard.balena-cloud.com/devices/eae2c9461555b2b5adda532ff1f04c6c

louisburton, 2022-12-28:
Hope you've have good holidays Zahari. 
Were you able to attempt reproduction on the devices above as I believe you reproduced the problem previously on these. Let me know if there's anything else I can provide @majorz. Thanks!

majorz, 2023-01-10:
I was able to find a way to reproduce this locally each time (after getting back from a long holiday). Starting to work on the patch now.

majorz, 2023-01-10:
The following PR fixes this (still in draft state): https://github.com/balena-os/meta-balena/pull/2963

I also created an issue upstream to hear the maintainers feedback for this: https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/1182

louisburton, 2023-01-11:
Thx!
Great to see the upstream fix also. Although would note CI is failing - not sure if that's a warning for this current draft PR.
https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/commit/686a5e792f6ac0f71347c836d490df43e949ceb3#note_1716991

Once rolled into an OS upgrade, happy for the devices linked above to be used to confirm issue, upgrade and confirm resolved.

majorz, 2023-01-11:
I've asked now whether the CI fail is related to the change. Awaiting on the answer and going to incorporate the upstream fix instead of mine if everything is alright.

louisburton, 2023-01-16:
Great news on the merge, thx for your efforts @majorz !
Do we have to wait for a NetworkManager release? I am wary that will take a further 2 weeks?
Can we perhaps do a test OS with a branch NM release that confirms you don't see evidence of reproduction on one of our test devices?
This issue has dragged a long time unfortunately and I'm wary to get a fix out asap.

majorz, 2023-01-16:
I included the upstream version of the patch in the related PR.

majorz, 2023-01-16:
Now awaiting on the CI checks to pass and will try to merge it asap, so that we get a new release soon as well.

louisburton, 2023-01-19:
@majorz The related PR also got merged yesterday kindly by Alex.

Does this mean we're able to now produce an OS release that includes this?

Can we please produce a release for the apolloOS, and could we then upgrade the above linked systems and confirm what was reproducible is no longer reproducible? 🙏  (it is less obvious for me to spot when iptables is corrupt):
https://github.com/balena-os/meta-balena/issues/2581#issuecomment-1360061901

Thanks!

closed 2023-01-24 by majorz