balena-os/meta-balena #2589: tests: hup: SIGABRT in cgo while loading local image

alexgg, 2022-04-15:
The following trace is seen since 9 days ago (6/04):

```
client_1  | [2022-04-15T17:57:09.984Z][worker-hup]         # ok - No under-voltage reports in the kernel logs
client_1  | [2022-04-15T17:57:20.754Z][worker-hup]         # Sending image to DUT
client_1  | [2022-04-15T17:57:20.754Z][worker-hup] image-98pr0hta
client_1  | [2022-04-15T17:57:28.567Z][worker-hup] /mnt/data/resin-data/image-98pr0hta
client_1  | [2022-04-15T17:57:28.681Z][worker-hup]         # Starting HUP
client_1  | [2022-04-15T17:57:38.793Z][worker-hup]         # Running: hostapp-update -f /mnt/data/resin-data/image-98pr0hta
client_1  | [2022-04-15T18:03:37.498Z][worker-hup]         not ok 1 - "hostapp-update -f /mnt/data/resin-data/image-98pr0hta" failed. stderr: init: dl-call-libc-early-init.c:37: _dl_call_libc_early_init: Assertion `sym != NULL' failed. SIGABRT: abort PC=0x84e453 m=0 sigcode=18446744073709551610 signal arrived during cgo execution  goroutine 1 [syscall, locked to thread]: runtime.cgocall(0x83a810, 0xc0001df5b8, 0xc0000be340)   /usr/lib/go/src/runtime/cgocall.go:154 +0x5b fp=0xc0001df588 sp=0xc0001df550 pc=0x40529b os/user._Cfunc_mygetpwuid_r(0x0, 0xc0001e7a70, 0x26ecbb0, 0x400, 0xc0000be340, 0x7fe000000000)   _cgo_gotypes.go:172 +0x48 fp=0xc0001df5b8 sp=0xc0001df588 pc=0x540b08 os/user.lookupUnixUid.func1.1(0x0, 0xc0001e7a70, 0xc0002ad730, 0xc0000be340, 0xc0001df658)   /usr/lib/go/src/os/user/cgo_lookup_unix.go:103 +0xd0 fp=0xc0001df608 sp=0xc0001df5b8 pc=0x542390 os/user.lookupUnixUid.func1(0x971d00)   /usr/lib/go/src/os/user/cgo_lookup_unix.go:103 +0x45 fp=0xc0001df640 sp=0xc0001df608 pc=0x542405 os/user.retryWithBuffer(0xc0002ad730, 0xc0001df718, 0xc0001df730, 0x40f35b)   /usr/lib/go/src/os/user/cgo_lookup_unix.go:247 +0x3e fp=0xc0001df698 sp=0xc0001df640 pc=0x541f7e os/user.lookupUnixUid(0x0, 0x0, 0x0, 0x0)   /usr/lib/go/src/os/user/cgo_lookup_unix.go:96 +0x132 fp=0xc0001df760 sp=0xc0001df698 pc=0x541392 os/user.current(0x447ecc, 0xcca3e0, 0x88f0000000000000)   /usr/lib/go/src/os/user/cgo_lookup_unix.go:49 +0x49 fp=0xc0001df7a0 sp=0xc0001df760 pc=0x540dc9 os/user.Current.func1()   /usr/lib/go/src/os/user/lookup.go:15 +0x25 fp=0xc0001df7c8 sp=0xc0001df7a0 pc=0x5420a5 sync.(*Once).doSlow(0xd05200, 0x9e7340)   /usr/lib/go/src/sync/once.go:68 +0xec fp=0xc0001df818 sp=0xc0001df7c8 pc=0x483b6c sync.(*Once).Do(...)   /usr/lib/go/src/sync/once.go:59 os/user.Current(0xc0002ee000, 0x7fe048111308, 0x8)   /usr/lib/go/src/os/user/lookup.go:15 +0x105 fp=0xc0001df848 sp=0xc0001df818 pc=0x540485 os/user.Lookup(0x9c656c, 0x6, 0x10, 0x95fe40, 0xd39f01)   /usr/lib/go/src/os/user/lookup.go:33 +0x26 fp=0xc0001df898 sp=0xc0001df848 pc=0x5404c6 github.com/docker/docker/pkg/chrootarchive.init.0()   /home/builder/go/pkg/mod/github.com/moby/moby@v17.12.0-ce-rc1.0.20200618181300-9dc6525e6118+incompatible/pkg/chrootarchive/archive.go:19 +0x36 fp=0xc0001df8f8 sp=0xc0001df898 pc=0x753c36 runtime.doInit(0xccd520)   /usr/lib/go/src/runtime/proc.go:6265 +0xec fp=0xc0001dfa48 sp=0xc0001df8f8 pc=0x447ecc runtime.doInit(0xccd6e0)   /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dfb98 sp=0xc0001dfa48 pc=0x447e52 runtime.doInit(0xcce900)   /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dfce8 sp=0xc0001dfb98 pc=0x447e52 runtime.doInit(0xccb340)   /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dfe38 sp=0xc0001dfce8 pc=0x447e52 runtime.doInit(0xccb440)   /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dff88 sp=0xc0001dfe38 pc=0x447e52 runtime.main()   /usr/lib/go/src/runtime/proc.go:208 +0x205 fp=0xc0001dffe0 sp=0xc0001dff88 pc=0x43a3c5 runtime.goexit()   /usr/lib/go/src/runtime/asm_amd64.s:1371 +0x1 fp=0xc0001dffe8 sp=0xc0001dffe0 pc=0x46d761  rax    0x0 rbx    0x26eb3c0 rcx    0x84e453 rdx    0x6 rdi    0x166c rsi    0x166c rbp    0x166c rsp    0x7fff08b542d0 r8     0x7 r9     0x26ed300 r10    0x8 r11    0x206 r12    0x6 r13    0xa7fd40 r14    0x25 r15    0x26eddd0 rip    0x84e453 rflags 0x206 cs     0x33 fs     0x0 gs     0x0, stdout: 21e904ae26cf Deleted: sha256:42e2693f2e62edc0563e43e891092168eb3b5101a4c6c691a8a87c72eeafac08 Deleted: sha256:cf770043e70a74ea332c6b2bb144ab2e4fd106b5d8f71619c8ca9a84bdd480a0 /mnt/sysroot/inactive/balena/tmp is not a mountpoint, code: 2
client_1  | [2022-04-15T18:03:37.499Z][worker-hup]           ---
client_1  | [2022-04-15T18:03:37.499Z][worker-hup]           stack: |
client_1  | [2022-04-15T18:03:37.499Z][worker-hup]             /mnt/sysroot/inactive/balena/tmp is not a mountpoint, code: 2
client_1  | [2022-04-15T18:03:37.500Z][worker-hup]                 retry.max_tries (lib/common/worker.js:308:12)
client_1  | [2022-04-15T18:03:37.500Z][worker-hup]                 runMicrotasks (<anonymous>)
client_1  |           test: Smoke test
client_1  | [2022-04-15T18:03:37.500Z][worker-hup]           ...
client_1  | [2022-04-15T18:03:37.516Z][worker-hup] 
client_1  | [2022-04-15T18:03:37.517Z][worker-hup]         Bail out! "hostapp-update -f /mnt/data/resin-data/image-98pr0hta" failed. stderr: init: dl-call-libc-early-init.c:37: _dl_call_libc_early_init: Assertion `sym != NULL' failed.
client_1  | [2022-04-15T18:03:37.531Z][worker-hup] Bail out! "hostapp-update -f /mnt/data/resin-data/image-98pr0hta" failed. stderr: init: dl-call-libc-early-init.c:37: _dl_call_libc_early_init: Assertion `sym != NULL' failed.
client_1  | [2022-04-15T18:03:37.532Z][worker-hup] Retrieving journalctl logs to the file /tmp/journalctl-7uognrnk.log ...
client_1  | [2022-04-15T18:03:37.536Z][worker-hup] Test suite completed. Tearing down now.
client_1  | [2022-04-15T18:03:37.536Z][worker-hup] Retrieving journalctl logs to the file /tmp/journalctl-24ouwi02.log ...
client_1  | [2022-04-15T18:03:47.696Z][worker-hup] Retrieving journalctl logs to the file /tmp/journalctl-3puivjlc.log ...
client_1  | [2022-04-15T18:03:57.860Z][worker-hup] Worker teardown
worker_1  | QEMU: powerOff
worker_1  | dnsmasq: exiting on receipt of SIGTERM
```

klutchell, 2022-04-19:
I just confirmed that this PR, for some reason, has broken local hostapp-update that we use in automated tests (`eg. hostapp-update -f balena-image.docker`)
https://github.com/balena-os/balena-intel/pull/456

When using an artifact from this commit, I am able to update:
- https://github.com/balena-os/balena-intel/commit/408b3d187fbf4c862d9ac0b26a65f971c283b400
- https://jenkins.product-os.io/view/Yocto/job/yocto-genericx86-64-ext/1312/

When using an artifact from this commit, the update fails with the SIGABRT mentioned above:
- https://github.com/balena-os/balena-intel/commit/d157e5d521e10ac84a69c7d97f16fede6d5b31cc
- https://jenkins.product-os.io/view/Yocto/job/yocto-genericx86-64-ext/1311/

It doesn't seem to matter what version we are coming from, only what version we are updating to.

I'm not sure where best to go next to narrow down the issue. Any ideas are appreciated, but this is currently blocking HUP tests on meta-balena PRs.

klutchell, 2022-04-19:
Here is the mentioned SIGABRT
```
++ /mnt/sysroot/inactive/current/boot/init -sysroot /mnt/sysroot/inactive
init: dl-call-libc-early-init.c:37: _dl_call_libc_early_init: Assertion `sym != NULL' failed.
SIGABRT: abort
PC=0x84e453 m=0 sigcode=18446744073709551610
signal arrived during cgo execution

goroutine 1 [syscall, locked to thread]:
runtime.cgocall(0x83a810, 0xc0001df5b8, 0xc0000be340)
    /usr/lib/go/src/runtime/cgocall.go:154 +0x5b fp=0xc0001df588 sp=0xc0001df550 pc=0x40529b
os/user._Cfunc_mygetpwuid_r(0x0, 0xc0001e5ad0, 0xec0bb0, 0x400, 0xc0000be340, 0x7f0100000000)
    _cgo_gotypes.go:172 +0x48 fp=0xc0001df5b8 sp=0xc0001df588 pc=0x540b08
os/user.lookupUnixUid.func1.1(0x0, 0xc0001e5ad0, 0xc0002ab730, 0xc0000be340, 0xc0001df658)
    /usr/lib/go/src/os/user/cgo_lookup_unix.go:103 +0xd0 fp=0xc0001df608 sp=0xc0001df5b8 pc=0x542390
os/user.lookupUnixUid.func1(0x971d00)
    /usr/lib/go/src/os/user/cgo_lookup_unix.go:103 +0x45 fp=0xc0001df640 sp=0xc0001df608 pc=0x542405
os/user.retryWithBuffer(0xc0002ab730, 0xc0001df718, 0xc0001df730, 0x40f35b)
    /usr/lib/go/src/os/user/cgo_lookup_unix.go:247 +0x3e fp=0xc0001df698 sp=0xc0001df640 pc=0x541f7e
os/user.lookupUnixUid(0x0, 0x0, 0x0, 0x0)
    /usr/lib/go/src/os/user/cgo_lookup_unix.go:96 +0x132 fp=0xc0001df760 sp=0xc0001df698 pc=0x541392
os/user.current(0x447ecc, 0xcca3e0, 0x88f0000000000000)
    /usr/lib/go/src/os/user/cgo_lookup_unix.go:49 +0x49 fp=0xc0001df7a0 sp=0xc0001df760 pc=0x540dc9
os/user.Current.func1()
    /usr/lib/go/src/os/user/lookup.go:15 +0x25 fp=0xc0001df7c8 sp=0xc0001df7a0 pc=0x5420a5
sync.(*Once).doSlow(0xd05200, 0x9e7340)
    /usr/lib/go/src/sync/once.go:68 +0xec fp=0xc0001df818 sp=0xc0001df7c8 pc=0x483b6c
sync.(*Once).Do(...)
    /usr/lib/go/src/sync/once.go:59
os/user.Current(0xc0002ec000, 0x7f01e5166418, 0x8)
    /usr/lib/go/src/os/user/lookup.go:15 +0x105 fp=0xc0001df848 sp=0xc0001df818 pc=0x540485
os/user.Lookup(0x9c656c, 0x6, 0x10, 0x95fe40, 0xd39f01)
    /usr/lib/go/src/os/user/lookup.go:33 +0x26 fp=0xc0001df898 sp=0xc0001df848 pc=0x5404c6
github.com/docker/docker/pkg/chrootarchive.init.0()
    /home/builder/go/pkg/mod/github.com/moby/moby@v17.12.0-ce-rc1.0.20200618181300-9dc6525e6118+incompatible/pkg/chrootarchive/archive.go:19 +0x36 fp=0xc0001df8f8 sp=0xc0001df898 pc=0x753c36
runtime.doInit(0xccd520)
    /usr/lib/go/src/runtime/proc.go:6265 +0xec fp=0xc0001dfa48 sp=0xc0001df8f8 pc=0x447ecc
runtime.doInit(0xccd6e0)
    /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dfb98 sp=0xc0001dfa48 pc=0x447e52
runtime.doInit(0xcce900)
    /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dfce8 sp=0xc0001dfb98 pc=0x447e52
runtime.doInit(0xccb340)
    /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dfe38 sp=0xc0001dfce8 pc=0x447e52
runtime.doInit(0xccb440)
    /usr/lib/go/src/runtime/proc.go:6242 +0x72 fp=0xc0001dff88 sp=0xc0001dfe38 pc=0x447e52
runtime.main()
    /usr/lib/go/src/runtime/proc.go:208 +0x205 fp=0xc0001dffe0 sp=0xc0001dff88 pc=0x43a3c5
runtime.goexit()
    /usr/lib/go/src/runtime/asm_amd64.s:1371 +0x1 fp=0xc0001dffe8 sp=0xc0001dffe0 pc=0x46d761
```

alexgg, 2022-04-20:
Could be a regression on glibc. Related items:
https://github.com/golang/go/issues/24787
https://github.com/golang/go/issues/13470
https://sourceware.org/bugzilla/show_bug.cgi?id=19341

Some components have stopped using `os/user` altogether like https://github.com/containerd/ttrpc/pull/13

klutchell, 2022-04-20:
discussion (restricted access):
https://www.flowdock.com/app/rulemotion/device-testing/threads/T6rqJwn2NlnxbTEyQLM9zJABTX8

commit 1a6dbc79 (2022-04-22): mobynit: Update to v0.2.0

Fixes #2589

Change-type: minor
PR #2598 (2022-04-22): mobynit: Update to v0.2.0
Fixes #2589

Change-type: minor

---
### Contributor checklist

- [x] Changes have been tested
  - [x] Covered in automated test suite
  - [ ] Manual test case recorded
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines)

### Reviewer Guidelines
- When submitting a review, please pick:
  - '*Approve*' if this change would be acceptable in the codebase (even if there are minor or cosmetic tweaks [... cut]
closed 2022-04-22 by klutchell