balena-os/meta-balena #2612: http proxy test fails on rpi4

floion, 2022-05-03:
```
# Subtest: Http-connect test
                # Creating redsocks.conf for http-connect...
                # Manually restarting services...
                ok 1 - Redsocks proxy service should be active
                # Getting proxy container logs...
                # Looking for http-connect connection logs...
                not ok 2 - www.google.com responded over http-connect proxy
                  ---
                  found: "2022/03/30 06:57:45 server.go:85: [socks5] 127.0.0.1:52316 <->
                    54.84.24.57:443 via DIRECT"
                  pattern: !re /\[http\] 127\.0\.0\.1:[0-9]* <->/
                  diff: >
                    --- expected

                    +++ actual

                    -/\[http\] 127\.0\.0\.1:[0-9]* <->/

                    +"2022/03/30 06:57:45 server.go:85: [socks5] 127.0.0.1:52316 <-> 54.84.24.57:443 via DIRECT"
```

Lately we have seen this test fail because at the second http-connect test we would still fetch the old socks5 logs from the glider container.

floion, 2022-05-03:
Proposed PR here: https://github.com/balena-os/meta-balena/pull/2563

klutchell, 2022-05-03:
Another proposed fix: https://github.com/balena-os/meta-balena/pull/2571

floion, 2022-05-08:
This issue is still there: https://jenkins.product-os.io/job/leviathan-v2-template/756/console

klutchell, 2022-05-09:
The error has changed now:
```
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.726Z][59415ad-os]                 # Looking for socks5 connection logs...
leviathan-v2-template-client-1  |                 not ok 2 - www.google.com responded over socks5 proxy
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.726Z][59415ad-os]                   ---
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.726Z][59415ad-os]                   found: "2022/05/06 19:01:06 server.go:105: [socks5] listening UDP on :8123"
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.727Z][59415ad-os]                   pattern: !re /\[socks5\] 127\.0\.0\.1:[0-9]* <->/
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.727Z][59415ad-os]                   diff: |
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.727Z][59415ad-os]                     --- expected
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.727Z][59415ad-os]                     +++ actual
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.727Z][59415ad-os]                     -/\[socks5\] 127\.0\.0\.1:[0-9]* <->/
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.728Z][59415ad-os]                     +"2022/05/06 19:01:06 server.go:105: [socks5] listening UDP on :8123"
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.728Z][59415ad-os]                   at:
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.728Z][59415ad-os]                     line: 158
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.728Z][59415ad-os]                     column: 13
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.728Z][59415ad-os]                     file: /data/suite/tests/connectivity/index.js
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.728Z][59415ad-os]                   stack: |
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.729Z][59415ad-os]                     /data/suite/tests/connectivity/index.js:158:13
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.731Z][59415ad-os]                   source: |
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.731Z][59415ad-os]                     							test.comment(`Looking for ${proxy} connection logs...`);
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.732Z][59415ad-os]                     							test.match(
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.732Z][59415ad-os]                     ------------^
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.732Z][59415ad-os]                     								proxyLog,
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.732Z][59415ad-os]                     								pattern,
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.732Z][59415ad-os]                   ...
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.748Z][59415ad-os] 
leviathan-v2-template-client-1  | [2022-05-06T19:02:05.749Z][59415ad-os]                 Bail out! www.google.com responded over socks5 proxy
```

closed 2022-05-10 by bulldozer-balena[bot]