balena-os/meta-balena #2634: tests: cloud: update preload test so strating up without internet is tested

rcooke-warwick, 2022-05-16:
The preload test should be updated to catch issues such as this one: 

https://github.com/balena-os/balena-supervisor/issues/1943

Where the device won't run the app if booted without internet connection. 

One option to do this is to do the preload test with wifi connect, although this won't work for emulated tests as the emulated devices don't have wifi

rcooke-warwick, 2022-05-16:
1 method to achieving this that would work for both emulated and real devices:

- Preload the image with an application (e.g hello world)
- flash the DUT
- disable the DUT's internet connection, but still maintain its local connection to the testbot. This can be done by disabling port forwarding on the testbot: `echo 0 > /proc/sys/net/ipv4/ip_forward`
- As you still have local connectivity to the DUT, SSH in and check that the preloaded app has started. In the case of the hello world app, you can see if the webserver is exposed on port 80 of the DUT as it is supposed to be,
- Afterwards, re-enable the DUT's internet connection by enabling port forwarding again `echo 1 > /proc/sys/net/ipv4/ip_forward`

rcooke-warwick, 2022-05-18:
We will limit this test to just use actual hardware, not emulated at the moment. This is due to difficulties disabling internet for the qemu DUT's with the current leviathan architecture

floion, 2022-05-30:
Hi @rcooke-warwick does this have a PR yet?

rcooke-warwick, 2022-05-30:
sorry @floion , I've been working on other things - I'll get the PR done today

rcooke-warwick, 2022-05-30:
@floion , this test should fail with the latest pi4 image if it is testing correctly?

rcooke-warwick, 2022-05-30:
PR is here: https://github.com/balena-os/meta-balena/pull/2651

closed 2022-06-20 by bulldozer-balena[bot]