balena-os/meta-balena #2679: tests: Error: All configured authentication methods failed

klutchell, 2022-06-13:
```
leviathan-v2-template-client-1  | [2022-06-13T10:09:57.583Z][worker-cloud] Test suite completed. Tearing down now.
leviathan-v2-template-client-1  | [2022-06-13T10:09:57.583Z][worker-cloud] Removing application testbot/5rraxd1w_MoveDevice
leviathan-v2-template-client-1  | [2022-06-13T10:10:10.451Z][worker-cloud] Retrieving journalctl logs to the file /tmp/journalctl-fng34n8z.log ...
leviathan-v2-template-client-1  | [2022-06-13T10:10:10.912Z][worker-cloud] Error: All configured authentication methods failed
leviathan-v2-template-client-1  |     at doNextAuth (/usr/app/node_modules/ssh2/lib/client.js:803:21)
leviathan-v2-template-client-1  |     at tryNextAuth (/usr/app/node_modules/ssh2/lib/client.js:993:7)
leviathan-v2-template-client-1  |     at tryNextAgentKey (/usr/app/node_modules/ssh2/lib/client.js:1002:11)
leviathan-v2-template-client-1  |     at USERAUTH_FAILURE (/usr/app/node_modules/ssh2/lib/client.js:366:20)
leviathan-v2-template-client-1  |     at 51 (/usr/app/node_modules/ssh2/lib/protocol/handlers.misc.js:337:16)
leviathan-v2-template-client-1  |     at Protocol.onPayload (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:2025:10)
leviathan-v2-template-client-1  |     at AESGCMDecipherBinding.decrypt (/usr/app/node_modules/ssh2/lib/protocol/crypto.js:1086:26)
leviathan-v2-template-client-1  |     at Protocol.parsePacket [as _parse] (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:1994:25)
leviathan-v2-template-client-1  |     at Protocol.parse (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:293:16)
leviathan-v2-template-client-1  |     at Socket.<anonymous> (/usr/app/node_modules/ssh2/lib/client.js:713:21)
leviathan-v2-template-client-1  |     at Socket.emit (events.js:400:28)
leviathan-v2-template-client-1  |     at addChunk (internal/streams/readable.js:293:12)
leviathan-v2-template-client-1  |     at readableAddChunk (internal/streams/readable.js:267:9)
leviathan-v2-template-client-1  |     at Socket.Readable.push (internal/streams/readable.js:206:10)
leviathan-v2-template-client-1  |     at TCP.onStreamRead (internal/stream_base_commons.js:188:23)
leviathan-v2-template-client-1  |     at TCP.callbackTrampoline (internal/async_hooks.js:130:17) {
leviathan-v2-template-client-1  |   level: 'client-authentication'
leviathan-v2-template-client-1  | }
leviathan-v2-template-client-1  | [2022-06-13T10:10:10.912Z][worker-cloud] Error while performing SSH command "journalctl --no-pager --no-hostname -a -b all", will retry...
leviathan-v2-template-client-1  | [2022-06-13T10:10:15.911Z][worker-cloud] Error: All configured authentication methods failed
leviathan-v2-template-client-1  |     at doNextAuth (/usr/app/node_modules/ssh2/lib/client.js:803:21)
leviathan-v2-template-client-1  |     at tryNextAuth (/usr/app/node_modules/ssh2/lib/client.js:993:7)
leviathan-v2-template-client-1  |     at tryNextAgentKey (/usr/app/node_modules/ssh2/lib/client.js:1002:11)
leviathan-v2-template-client-1  |     at USERAUTH_FAILURE (/usr/app/node_modules/ssh2/lib/client.js:366:20)
leviathan-v2-template-client-1  |     at 51 (/usr/app/node_modules/ssh2/lib/protocol/handlers.misc.js:337:16)
leviathan-v2-template-client-1  |     at Protocol.onPayload (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:2025:10)
leviathan-v2-template-client-1  |     at AESGCMDecipherBinding.decrypt (/usr/app/node_modules/ssh2/lib/protocol/crypto.js:1086:26)
leviathan-v2-template-client-1  |     at Protocol.parsePacket [as _parse] (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:1994:25)
leviathan-v2-template-client-1  |     at Protocol.parse (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:293:16)
leviathan-v2-template-client-1  |     at Socket.<anonymous> (/usr/app/node_modules/ssh2/lib/client.js:713:21)
leviathan-v2-template-client-1  |     at Socket.emit (events.js:400:28)
leviathan-v2-template-client-1  |     at addChunk (internal/streams/readable.js:293:12)
leviathan-v2-template-client-1  |     at readableAddChunk (internal/streams/readable.js:267:9)
leviathan-v2-template-client-1  |     at Socket.Readable.push (internal/streams/readable.js:206:10)
leviathan-v2-template-client-1  |     at TCP.onStreamRead (internal/stream_base_commons.js:188:23)
leviathan-v2-template-client-1  |     at TCP.callbackTrampoline (internal/async_hooks.js:130:17) {
leviathan-v2-template-client-1  |   level: 'client-authentication'
leviathan-v2-template-client-1  | }
```

rcooke-warwick, 2022-06-13:
@klutchell where were you seeing this? I thought this was solved last week via https://github.com/balena-os/leviathan/pull/768

klutchell, 2022-06-13:
@rcooke-warwick https://jenkins.product-os.io/job/leviathan-v2-template/1826/console

closed 2022-07-20 by bulldozer-balena[bot]