balena-os/meta-balena #2744: not ok 4 - Local SSH authentication without custom keys is not allowed in production mode

klutchell, 2022-07-25:
```
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:05:44.537Z][65a9b86-cloud]             ok 3 - Should wait for os-sshkeys.service to be active
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:05:49.116Z][65a9b86-cloud] Connection lost before handshake
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:05:53.428Z][65a9b86-cloud] Connection lost before handshake
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:05:57.933Z][65a9b86-cloud] Connection lost before handshake
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:02.538Z][65a9b86-cloud] Connection lost before handshake
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.938Z][65a9b86-cloud] Connection lost before handshake
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.955Z][65a9b86-cloud]             not ok 4 - Local SSH authentication without custom keys is not allowed in production mode
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.955Z][65a9b86-cloud]               ---
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.955Z][65a9b86-cloud]               diff: |
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.956Z][65a9b86-cloud]                 --- expected
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.956Z][65a9b86-cloud]                 +++ actual
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.956Z][65a9b86-cloud]                 @@ -1,1 +1,1 @@
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.956Z][65a9b86-cloud]                 -/All configured authentication methods failed/
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.958Z][65a9b86-cloud]                 +"Error: Connection lost before handshake"
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.958Z][65a9b86-cloud]               pattern: !re /All configured authentication methods failed/
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.959Z][65a9b86-cloud]               at:
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.959Z][65a9b86-cloud]                 line: 118
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.959Z][65a9b86-cloud]                 column: 19
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.959Z][65a9b86-cloud]                 file: /data/suite/tests/ssh-auth/index.js
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.959Z][65a9b86-cloud]               stack: |
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.959Z][65a9b86-cloud]                 /data/suite/tests/ssh-auth/index.js:118:19
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.959Z][65a9b86-cloud]                 /data/suite/tests/ssh-auth/index.js:111:6
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.960Z][65a9b86-cloud]               source: |
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.960Z][65a9b86-cloud]                 					}).catch((err) => {
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.960Z][65a9b86-cloud]                 						return test.match(
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.960Z][65a9b86-cloud]                 ------------------^
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.960Z][65a9b86-cloud]                 							err.message,
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.960Z][65a9b86-cloud]                 							/All configured authentication methods failed/,
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.960Z][65a9b86-cloud]               ...
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.975Z][65a9b86-cloud] 
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.976Z][65a9b86-cloud]             Bail out! Local SSH authentication without custom keys is not allowed in production mode
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.981Z][65a9b86-cloud] Bail out! Local SSH authentication without custom keys is not allowed in production mode
jenkins-leviathan-v2-template-3196-client-1  | [2022-07-24T20:06:06.982Z][65a9b86-cloud] Test suite completed. Tearing down now
```
[consoleText3.txt](https://github.com/balena-os/meta-balena/files/9181772/consoleText3.txt)
https://jenkins.product-os.io/job/leviathan-v2-template/3196

klutchell, 2022-07-25:
@jakogut  First observed on Intel NUC but it seems that any device that returns an error other than `All configured authentication methods failed` will fail here since we expect and error and inspect the output of the first response. Given a few more attempts the correct error may have been returned.

rcooke-warwick, 2022-08-10:
@klutchell I've got a fix that I'm just testing - first a fix for this error, and then a fix for the error that comes after this part of the test when it continues past this point. (both these errors are for physical DUT only)

closed 2022-08-16 by bulldozer-balena[bot]