balena-os/meta-balena #2799: not ok 4 - Should wait for config-json.service to be inactive

jakogut, 2022-09-21:
```
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.650Z][worker-cloud]                 not ok 4 - Should wait for config-json.service to be inactive
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.650Z][worker-cloud]                   ---
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.650Z][worker-cloud]                   at:
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.651Z][worker-cloud]                     line: 74
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.651Z][worker-cloud]                     column: 13
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.651Z][worker-cloud]                     file: /data/suite/tests/ssh-auth/index.js
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.651Z][worker-cloud]                   found:
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.652Z][worker-cloud]                     !error
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.652Z][worker-cloud]                     name: Error
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.652Z][worker-cloud]                     message: |-
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.652Z][worker-cloud]                       Condition async () => {
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.652Z][worker-cloud]                                   return this.cloud
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.653Z][worker-cloud]                                     .executeCommandInHostOS(
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.653Z][worker-cloud]                                       `systemctl is-active ${serviceName} || true`,
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.653Z][worker-cloud]                                       target,
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.653Z][worker-cloud]                                     )
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.653Z][worker-cloud]                                     .then((serviceStatus) => {
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.654Z][worker-cloud]                                       return Promise.resolve(serviceStatus === state);
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.655Z][worker-cloud]                                     })
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.655Z][worker-cloud]                                     .catch((err) => {
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.655Z][worker-cloud]                                       Promise.reject(err);
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.655Z][worker-cloud]                                     });
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.655Z][worker-cloud]                                 } timed out
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.655Z][worker-cloud]                     stack: |-
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.656Z][worker-cloud]                       Error: Condition async () => {
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.656Z][worker-cloud]                                   return this.cloud
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.657Z][worker-cloud]                                     .executeCommandInHostOS(
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.657Z][worker-cloud]                                       `systemctl is-active ${serviceName} || true`,
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.657Z][worker-cloud]                                       target,
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.657Z][worker-cloud]                                     )
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.657Z][worker-cloud]                                     .then((serviceStatus) => {
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.658Z][worker-cloud]                                       return Promise.resolve(serviceStatus === state);
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.658Z][worker-cloud]                                     })
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.658Z][worker-cloud]                                     .catch((err) => {
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.658Z][worker-cloud]                                       Promise.reject(err);
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.658Z][worker-cloud]                                     });
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.659Z][worker-cloud]                                 } timed out
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.659Z][worker-cloud]                           at _waitUntil (/usr/app/lib/common/utils.js:103:11)
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.659Z][worker-cloud]                           at _waitUntil (/usr/app/lib/common/utils.js:118:11)
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.659Z][worker-cloud]                           at Object.waitUntil (/usr/app/lib/common/utils.js:121:3)
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.660Z][worker-cloud]                   source: |
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.660Z][worker-cloud]                     		}).then(() => {
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.660Z][worker-cloud]                     			return t.resolves(
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.660Z][worker-cloud]                     ------------^
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.660Z][worker-cloud]                     				that.waitForServiceState(
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.661Z][worker-cloud]                     					'config-json.service',
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.661Z][worker-cloud]                   ...
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.677Z][worker-cloud] 
jenkins-leviathan-v2-template-5155-client-1  | [2022-09-21T21:03:45.678Z][worker-cloud]                 Bail out! Should wait for config-json.service to be inactive
```

jakogut, 2022-10-07:
@alexgg This is one of the main blockers for CI right now, this is stopping the kirkstone branches from being merged for x86 device types, among other PRs

jakogut, 2022-10-07:
https://github.com/balena-os/balena-generic/pull/134

rcooke-warwick, 2022-10-27:
@jakogut @alexgg I'll update this issue as there's a lot of context on zulip rn that is missing from here:

This test was failing predominantly (afaik) on device repo PR's where we were shifting to kirkstone eg. https://github.com/balena-os/balena-intel/pull/560

The test was failing, because SSH access via balena cloud proxy is broken for images built from these PR's when the device is in production mode. @alexgg is currently investigating why.