balena-os/meta-balena #2816: tests/hup: Hostapp update suite: Cannot read property 'replace' of null

alexgg, 2022-10-10:
```
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.251Z][worker-hup] # Subtest: Hostapp update suite
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.252Z][worker-hup]     not ok 1 - Cannot read property 'replace' of null
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.253Z][worker-hup]       ---
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.253Z][worker-hup]       stack: |
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.253Z][worker-hup]         BalenaSDK.fetchOS (lib/components/balena/sdk.js:324:21)
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.253Z][worker-hup]         Proxy.run (/data/suite/suite.js:254:14)
jenkins-leviathan-v2-template-5566-client-1  |       at:
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.253Z][worker-hup]         line: 324
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.254Z][worker-hup]         column: 21
jenkins-leviathan-v2-template-5566-client-1  |         file: lib/components/balena/sdk.js
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.254Z][worker-hup]         function: BalenaSDK.fetchOS
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.254Z][worker-hup]       type: TypeError
jenkins-leviathan-v2-template-5566-client-1  |       test: Hostapp update suite
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.254Z][worker-hup]       source: >
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.254Z][worker-hup]         		// if prod variant is still present after being normalized, replace it with dev
jenkins-leviathan-v2-template-5566-client-1  |         		version = version.replace('.prod', '.dev');
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.254Z][worker-hup]         --------------------^
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.254Z][worker-hup]         		const path = join(
jenkins-leviathan-v2-template-5566-client-1  |       ...
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.264Z][worker-hup] 
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.265Z][worker-hup]     Bail out! Cannot read property 'replace' of null
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.270Z][worker-hup] Bail out! Cannot read property 'replace' of null
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.270Z][worker-hup] Test suite completed. Tearing down now.
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.274Z][worker-hup] Removing downloads directory...
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.274Z][worker-hup] Removing npm dependencies for suite:
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.317Z][worker-hup] Teardown complete.
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.318Z][worker-hup] Test Finished
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.318Z][worker-hup] Suite run complete
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.318Z][worker-hup] Exiting suite child process...
jenkins-leviathan-v2-template-5566-core-1    | exit
jenkins-leviathan-v2-template-5566-core-1    | Suite exit code is: 1
jenkins-leviathan-v2-template-5566-client-1  | [2022-10-10T10:00:49.326Z][worker-hup] Test suite has exited with: FAIL
```

alexgg, 2022-10-10:
Might be caused by not having a final release for `generic-amd64` yet.

rcooke-warwick, 2022-10-13:
@alexgg most certainly this is the case ^

error is coming from here: https://github.com/balena-os/leviathan/blob/master/core/lib/components/balena/sdk.js#L316

the sdk is being used to fetch the most recent version of the device type, and must be returning null when it doesn't find a release for it - leading to `Cannot read property 'replace' of null`

rcooke-warwick, 2022-10-14:
cc @vipulgupta2048 this is the issue you mentioned yesterday too

alexgg, 2022-10-15:
@rcooke-warwick @vipulgupta2048 the reason why this is a problem is that the HUP test will always fail for the first final release, so we will never get to automatically deploy the first release. The HUP test should bail out if there is no previous release, not fail.

vipulgupta2048, 2022-10-16:
@alexgg Thanks for opening this issue. To fix this,  I had the exact same solution in mind. I will get it done. The only speedbump I see is bailing or skipping out of tests could be a slippery slope. The test suite might get skipped incorrectly when we don't find any releases for a device type. I need to make the fix in a such a way that it only skips the HUP suite when 
1. A device type is found but 
2. No release is found for the deviceType 
3. I wonder if that's even possible. 

Side question: Would it be so bad if we just skipped HUP anytime we don't find releases for a device type? The slippery slope in this case being that HUP could get skipped if there is ever an invalid device type that gets tested.

closed 2022-11-04 by bulldozer-balena[bot]