balena-os/meta-balena #2823: Chronyd healthcheck not notifying systemd watchdog

alexgg, 2022-10-18:
The chronyd healthcheck introduced in v2.98.45 is not notifying systemd correctly and the service is being restarted every 4 minutes.
```
systemd[1]: chronyd.service: Got notification message from PID 26321, but reception only permitted for main PID
```

alexgg, 2022-10-19:
On first look it's not a problem with chronyd but with the healthdog binary. Systemd does not receive any watchdog signal. Also, the test framework needs a test to make sure the healthchecks are working which we don't currently have.

I tested v2.99.19 on a RPI4 which already contains the chronyd healthcheck changes and that works fine, while master does not. I suspect the Rust version update.

alexgg, 2022-10-19:
Test plan:

v2.101.0: Should work
v2.101.1: healthdog update to v1.0.6
v2.101.11: Should work if v2.101.1 did
v2.102.0: Rust update

alexgg, 2022-10-19:
The conclusion of the investigation is that when using chrony 4.2 healthdog is not able to send watchdog events to systemd so the service is restarted every 4m when the timeout expires.

Functionally, restarting chronyd should not have other implications, but according to reports some timers might be affected when the time is abruptly stepped by chrony restarting instead of synched by a running service.

nbmoa, 2022-10-19:
Just as a confirmation.
We experience a less stable timer behaviour with the latest balena release, resulting in occasionally early timer events (probably related with syncing time over a wifi/modem connection).
Thanks for addressing it

closed 2022-10-19 by bulldozer-balena[bot]

alexgg, 2022-10-28:
(internal) https://balena.zulipchat.com/#narrow/stream/345889-loop.2Fbalena-os/topic/Known.20issue.20for.20releases.20.3E.3D2.2E103.2E0

The root of this are the security restrictions introduced in https://github.com/mlichvar/chrony/commit/83f96efdfd2d42a8de51ac3b05120acf5292bb00

And in particular the removal of the `CAP_SYS_ADMIN` capability that affects the way healthdog uses `execve` to become chronyd.

The following change also addresses the problem:
```
--- chronyd.service
+++ chronyd.service.orig
@@ -13,7 +13,7 @@

 CapabilityBoundingSet=~CAP_AUDIT_CONTROL CAP_AUDIT_READ CAP_AUDIT_WRITE
 CapabilityBoundingSet=~CAP_BLOCK_SUSPEND CAP_KILL CAP_LEASE CAP_LINUX_IMMUTABLE
-CapabilityBoundingSet=~CAP_MAC_ADMIN CAP_MAC_OVERRIDE CAP_MKNOD
+CapabilityBoundingSet=~CAP_MAC_ADMIN CAP_MAC_OVERRIDE CAP_MKNOD CAP_SYS_ADMIN
 CapabilityBoundingSet=~CAP_SYS_BOOT CAP_SYS_CHROOT CAP_SYS_MODULE CAP_SYS_PACCT
 CapabilityBoundingSet=~CAP_SYS_PTRACE CAP_SYS_RAWIO CAP_SYS_TTY_CONFIG CAP_WAKE_ALARM
 DeviceAllow=char-pps rw
```
Because CAP_SYS_ADMIN has a broad range of privileges and I am not that familiar with healthdog it's not clear to me whether it's preferable to add a `NotifyAccess=exec` or to re-instate this privilege. I tend to think `NotifyAccess=exec` is more secure, assuming we can test that heathdog still works.

PR #2824 (2022-10-19): Revert "chrony: update to version 4.1 to match kirkstone's version"
With the updated version the systemd watchdog events are not received.

Fixes #2823

This reverts commit 5049451b32d2d2c59f2aeb4713b2e6539bffeb49.

Change-type: patch

---
### Contributor checklist

- [ ] Changes have been tested
  - [ ] Covered in automated test suite
  - [ ] Manual test case recorded
- [ ] `Change-type` present on at least one commit
- [ ] `Signed-off-by` is present
- [ ] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines) [... cut]
commit 3ea5184c (2022-10-19): Revert "chrony: update to version 4.1 to match kirkstone's version"

With the updated version the systemd watchdog events are not received.

Fixes #2823

This reverts commit 5049451b32d2d2c59f2aeb4713b2e6539bffeb49.

Change-type: patch