balena-os/meta-balena #2834: Supervisor may need manual restart after being killed by os-config during startup

acostach, 2022-10-27:
The issue was reproduced 5x times on a Jetson Xavier Orin NX in the staging environment. The build is based on Kirkstone.

Supervisor had to be restarted manually for the device to register in the dashboard. This happens on the first boot, right after provisioning, and appears to be sporadic. 

```
root@30d13df:~# systemctl status balena-supervisor
* balena-supervisor.service - Balena supervisor
     Loaded: loaded (8;;file://30d13df/lib/systemd/system/balena-supervisor.service/lib/systemd/system/balena-supervisor.service8;;; enabled;
vendor preset: enabled)8;;
    Drop-In: /etc/systemd/system/balena-supervisor.service.d
             `-8;;file://30d13df/etc/systemd/system/balena-supervisor.service.d/balena-supervisor-conf.confbalena-supervisor-conf.conf8;;8;;
     Active: inactive (dead)
root@30d13df:~# journalctl | grep supervisor
Apr 28 17:42:26 localhost bindmount[1710]: INFO: Bindmounting /etc/balena-supervisor in /mnt/state/root-overlay/etc/balena-supervisor ...
Apr 28 17:42:26 localhost bindmount[1710]: WARN: /etc/balena-supervisor is not an empty entry. You are going to shadow content.
Apr 28 17:42:26 localhost bindmount[1710]: INFO: Successfully mounted /mnt/state/root-overlay/etc/balena-supervisor.
Apr 28 17:42:30 localhost os-config-json[2672]: [os-config-json][INFO] New balena-supervisor configuration:
Apr 28 17:42:31 localhost echo[2795]: balena-supervisor configuration changed
Oct 26 09:24:04 30d13df os-config[2857]: Stopping balena-supervisor.service...
Oct 26 09:24:04 30d13df systemctl[2800]: Job for balena-supervisor.service canceled.
Oct 26 09:24:04 30d13df systemd[1]: balena-supervisor-conf.service: Main process exited, code=exited, status=1/FAILURE
Oct 26 09:24:04 30d13df systemd[1]: balena-supervisor-conf.service: Failed with result 'exit-code'.
Oct 26 09:24:04 30d13df os-config[2857]: Awaiting balena-supervisor.service to exit...
Oct 26 09:24:04 30d13df systemd[1]: Failed to start balena-supervisor.json watcher service.
root@30d13df:~# systemctl restart balena-supervisor
root@30d13df:~# journalctl | grep supervisor
Apr 28 17:42:26 localhost bindmount[1710]: INFO: Bindmounting /etc/balena-supervisor in /mnt/state/root-overlay/etc/balena-supervisor ...
Apr 28 17:42:26 localhost bindmount[1710]: WARN: /etc/balena-supervisor is not an empty entry. You are going to shadow content.
Apr 28 17:42:26 localhost bindmount[1710]: INFO: Successfully mounted /mnt/state/root-overlay/etc/balena-supervisor.
Apr 28 17:42:30 localhost os-config-json[2672]: [os-config-json][INFO] New balena-supervisor configuration:
Apr 28 17:42:31 localhost echo[2795]: balena-supervisor configuration changed
Oct 26 09:24:04 30d13df os-config[2857]: Stopping balena-supervisor.service...
Oct 26 09:24:04 30d13df systemctl[2800]: Job for balena-supervisor.service canceled.
Oct 26 09:24:04 30d13df systemd[1]: balena-supervisor-conf.service: Main process exited, code=exited, status=1/FAILURE
Oct 26 09:24:04 30d13df systemd[1]: balena-supervisor-conf.service: Failed with result 'exit-code'.
Oct 26 09:24:04 30d13df os-config[2857]: Awaiting balena-supervisor.service to exit...
Oct 26 09:24:04 30d13df systemd[1]: Failed to start balena-supervisor.json watcher service.
Oct 26 09:43:59 30d13df balena-supervisor[4490]: Error response from daemon: No such container: resin_supervisor
Oct 26 09:43:59 30d13df balena-supervisor[4501]: Error response from daemon: No such container: balena_supervisor
Oct 26 09:43:59 30d13df balena-supervisor[4512]: active
Oct 26 09:43:59 30d13df balena-supervisor[4542]: Error: No such object: balena_supervisor
Oct 26 09:43:59 30d13df c785186a190d[2884]: find: /mnt/root/tmp/balena-supervisor/services: No such file or directory
Oct 26 09:43:59 30d13df balena-supervisor[4581]: find: /mnt/root/tmp/balena-supervisor/services: No such file or directory
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [info]    Supervisor v14.2.10 starting up...
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [info]    Setting host to discoverable
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [debug]   Starting systemd unit: avahi-daemon.service
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [debug]   Starting systemd unit: avahi-daemon.socket
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [debug]   Starting logging infrastructure
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [info]    Starting firewall
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [warn]    Invalid firewall mode: . Reverting to state: off
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [info]    Applying firewall mode: off
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [success] Firewall mode applied
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [debug]   Starting api binder
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [debug]   Performing database cleanup for container log timestamps
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [info]    Previous engine snapshot was not stored. Skipping cleanup.
Oct 26 09:44:01 30d13df balena-supervisor[4581]: [debug]   Handling of local mode switch is completed
Oct 26 09:44:01 30d13df balena-supervisor[4581]: (node:1) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
```

Today, further attempts to reproduce the issue failed because the SV was restarted automatically and registered in the dashboard.

```
root@27fbd96:~# journalctl | grep superviso
Apr 28 17:42:26 localhost bindmount[1712]: INFO: Bindmounting /etc/balena-supervisor in /mnt/state/root-overlay/etc/balena-supervisor ...
Apr 28 17:42:26 localhost bindmount[1712]: WARN: /etc/balena-supervisor is not an empty entry. You are going to shadow content.
Apr 28 17:42:26 localhost bindmount[1712]: INFO: Successfully mounted /mnt/state/root-overlay/etc/balena-supervisor.
Apr 28 17:42:31 localhost os-config-json[2728]: [os-config-json][INFO] New balena-supervisor configuration:
Apr 28 17:42:31 localhost echo[2827]: balena-supervisor configuration changed
Oct 27 10:56:32 27fbd96 os-config[2803]: Stopping balena-supervisor.service...
Oct 27 10:56:32 27fbd96 systemctl[2835]: Job for balena-supervisor.service canceled.
Oct 27 10:56:32 27fbd96 systemd[1]: balena-supervisor-conf.service: Main process exited, code=exited, status=1/FAILURE
Oct 27 10:56:32 27fbd96 systemd[1]: balena-supervisor-conf.service: Failed with result 'exit-code'.
Oct 27 10:56:32 27fbd96 os-config[2803]: Awaiting balena-supervisor.service to exit...
Oct 27 10:56:32 27fbd96 systemd[1]: Failed to start balena-supervisor.json watcher service.
Oct 27 10:56:32 27fbd96 os-config[2803]: Starting balena-supervisor.service...
Oct 27 10:56:34 27fbd96 balena-supervisor[3248]: Error response from daemon: No such container: resin_supervisor
Oct 27 10:56:34 27fbd96 balena-supervisor[3260]: Error response from daemon: No such container: balena_supervisor
Oct 27 10:56:34 27fbd96 balena-supervisor[3271]: active
Oct 27 10:56:34 27fbd96 balena-supervisor[3302]: Error: No such object: balena_supervisor
Oct 27 10:56:36 27fbd96 5df59d2f71dd[2856]: find: /mnt/root/tmp/balena-supervisor/services: No such file or directory
Oct 27 10:56:36 27fbd96 balena-supervisor[3340]: find: /mnt/root/tmp/balena-supervisor/services: No such file or directory
Oct 27 10:56:38 27fbd96 balena-supervisor[3340]: [info]    Supervisor v14.2.10 starting up...
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [info]    Setting host to discoverable
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [debug]   Starting systemd unit: avahi-daemon.service
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [debug]   Starting systemd unit: avahi-daemon.socket
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [debug]   Starting logging infrastructure
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [info]    Starting firewall
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [warn]    Invalid firewall mode: . Reverting to state: off
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [info]    Applying firewall mode: off
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [success] Firewall mode applied
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [debug]   Starting api binder
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [debug]   Performing database cleanup for container log timestamps
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [info]    Previous engine snapshot was not stored. Skipping cleanup.
Oct 27 10:56:39 27fbd96 balena-supervisor[3340]: [debug]   Handling of local mode switch is completed

```

Internal discussion: https://balena.zulipchat.com/#narrow/stream/345889-loop.2Fbalena-os/topic/supervisor-startup-failure/near/306203131

@majorz investigated the device and noted that this is a racing issue in the service start sequence which is not device-related

alexgg, 2022-11-07:
@majorz for me the open question here, if os-config is killing balena-supervisor, why is it not starting it? I think the logs above don't show the full picture as we see (before the restart):
```
Apr 28 17:42:31 localhost echo[2795]: balena-supervisor configuration changed
Oct 26 09:24:04 30d13df os-config[2857]: Stopping balena-supervisor.service...
Oct 26 09:24:04 30d13df systemctl[2800]: Job for balena-supervisor.service canceled.
Oct 26 09:24:04 30d13df systemd[1]: balena-supervisor-conf.service: Main process exited, code=exited, status=1/FAILURE
Oct 26 09:24:04 30d13df systemd[1]: balena-supervisor-conf.service: Failed with result 'exit-code'.
Oct 26 09:24:04 30d13df os-config[2857]: Awaiting balena-supervisor.service to exit...
Oct 26 09:24:04 30d13df systemd[1]: Failed to start balena-supervisor.json watcher service.
```
So here is what I read from that:

* There is a configuration change to balena-supervisor, so the unit configuration file restarts it
* In the mean time the `os-config join` command attempts to stop the balena-supervisor, and then waits in `await_service_exit_impl`
* The "Timed out awaiting service to exit" is not printed, but neither is the service restarted.

I think we need to make this more robust in `os-config`.

alexgg, 2022-11-14:
Fixed in v2.106.5

closed 2022-11-14 by alexgg