balena-os/meta-balena #2892: Replace archival of logs command with archiveLogs helper

vipulgupta2048, 2022-11-30:
With the merge of https://github.com/balena-os/leviathan/pull/853, @vipulgupta2048 needs to replace custom logic to pull journalctl logs in the test suites with this helper.

vipulgupta2048, 2022-12-01:
@klutchell in your improvement PR of #2893, can you replace this helper as well in the HUP suite ^
one less PR to make.

klutchell, 2022-12-01:
> @klutchell in your improvement PR of #2893, can you replace this helper as well in the HUP suite ^ one less PR to make.

What needs to be replaced in the HUP suite? It's already using the helper, it just also includes a custom command parameter.

https://github.com/balena-os/meta-balena/blob/master/tests/suites/hup/suite.js#L164-L170

vipulgupta2048, 2022-12-01:
@klutchell I think the same test suite has two teardowns: https://github.com/balena-os/meta-balena/blob/master/tests/suites/hup/suite.js#L347  (one for initDUT and other one for suite end) 

Since the custom command is now the default command of the helper, then why not use the helper everywhere is what I mean

klutchell, 2022-12-01:
So you mean just remove the 3rd custom command arg from https://github.com/balena-os/meta-balena/blob/master/tests/suites/hup/suite.js#L164-L170?

vipulgupta2048, 2022-12-01:
Yeap

klutchell, 2022-12-01:
That change can't go in https://github.com/balena-os/meta-balena/pull/2893 because the Leviathan change to the helper [hasn't merged to meta-balena yet](https://github.com/balena-os/meta-balena/pull/2886), so if I include the change it would break log collection for HUP.