balena-os/meta-balena #2907: DNS servers via DHCP are not being picked up.

vinceve, 2022-12-07:
We're currently deploying an intel NUC at a customer of ours and they are using internal DNS servers. The google DNS 8.8.8.8 is blocked on the network. When we flash the image, we're seeing that the DNS servers are not being picked up. I'm seeing in `/etc/resolv.dnsmasq` that it contains `nameserver 8.8.8.8`. When I change it to one of the internal DNS servers it resolves and the nuc comes online in the Balena platform.

I tried setting the dnsServers setting to `"null"` in the config.json and even configuring manually the internal DNS, but nothing seems to work. According to the documentation, it should work out of the box, if I understand correctly.

majorz, 2022-12-08:
Looking into this now.

majorz, 2022-12-08:
I tried this with a few OS versions, but i always get the proper DNS servers coming from DHCP inside the `/etc/resolv.dnsmasq`.

Can you please run the following command and paste the output (you may need to change eth0 to the appropriate interface):
`nmcli -f ip4.dns device show eth0`

Also can you please paste here the contents of the following files:
/var/run/resolvconf/interface/NetworkManager
/var/run/NetworkManager/resolv.conf
/var/run/NetworkManager/no-stub-resolv.conf

Also the contents of the following command:
`journalctl | grep nameserver`

nojocodex, 2022-12-09:
I think I may have run into this as well.

It seems `/sbin/resolvconf` fails because `/lib/normalize-resolvconf` doesn't exist. Here's a snippet from `journalctl`:

```
NetworkManager: <info>  [1670578124.4021] device (eth0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
NetworkManager: <info>  [1670578124.4023] manager: NetworkManager state is now CONNECTED_LOCAL
NetworkManager: <info>  [1670578124.4025] manager: NetworkManager state is now CONNECTED_SITE
NetworkManager: <info>  [1670578124.4025] policy: set 'Wired connection 1' (eth0) as default for IPv4 routing and DNS
NetworkManager: <info>  [1670578124.4027] dns-mgr: Writing DNS information to /sbin/resolvconf
NetworkManager: /sbin/resolvconf: line 173: /lib/resolvconf/normalize-resolvconf: No such file or directory
NetworkManager: <warn>  [1670578124.4066] dns-mgr: resolvconf failed with status 32512
NetworkManager: <warn>  [1670578124.4068] dns-mgr: could not commit DNS changes: resolvconf failed with status 32512
NetworkManager: <info>  [1670578124.4068] device (eth0): Activation: successful, device activated.
NetworkManager: <info>  [1670578124.4072] manager: startup complete
```

The file `/lib/normalize-resolvconf` was split out in `resolvconf` upstream [here](https://salsa.debian.org/debian/resolvconf/-/commit/ce9f98a6f8eb38a5508435cce4170c30d0d52b80), which broke Yocto. Yocto has a fix upstream [here](https://github.com/openembedded/openembedded-core/commit/1b0581fd241cc9de2feda896aefbf055dc0099dc).

majorz, 2022-12-09:
@nojocodex great find! It makes sense for resolveconf to be the problem as it is writing the entries in `/etc/resolv.dnsmasq`. We will look into this.

vinceve, 2022-12-09:
@majorz I will provide you with the requested information above.

majorz, 2022-12-09:
@vinceve thanks, but for now I won't need it as I was able to reproduce the problem locally.

commit bf8ad3a9 (2022-12-15): resolvconf: Add upstream 1.91 recipe for kirkstone

Current resolvconf version is broken for kirkstone, which is fixed upstream,
so we reuse the fixed version from upstream.

The symptom is that DNS servers provided by DHCP are not being used.

Closes #2907

Change-type: patch
Changelog-entry: Add upstream resolvconf 1.91 recipe for kirkstone
commit af89ddb5 (2022-12-16): resolvconf: Add upstream 1.91 recipe for kirkstone

Current resolvconf version is broken for kirkstone, which is fixed upstream,
so we reuse the fixed version from upstream.

The symptom is that DNS servers provided by DHCP are not being used.

Closes #2907

Change-type: patch
Changelog-entry: Add upstream resolvconf 1.91 recipe for kirkstone
PR #2924 (2022-12-16): resolvconf: Add upstream 1.91 recipe for kirkstone
Current resolvconf version is broken for kirkstone, which is fixed upstream, so we reuse the fixed version from upstream.

The symptom is that DNS servers provided by DHCP are not being used.

Closes #2907

Change-type: patch
Changelog-entry: Add upstream resolvconf 1.91 recipe for kirkstone

---
### Contributor checklist

- [x] Changes have been tested
  - [ ] Covered in automated test suite
  - [ ] Manual test case recorded
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Ope [... cut]
closed 2022-12-16 by balena-ci