balena-os/meta-balena #2964: Networkmanager 1.40.4 does not include connectivity check?

everhardt, 2023-01-11:
When I run balenaOS 2.108.1 and set the log level for connectivity checking in the NetworkManager to debug (`nmcli g log level DEBUG domains CONCHECK`), I see the following log lines in journalctl:

```
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9106] connectivity: (wwan0,IPv6,38) check completed: FAKE; fake result
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9105] connectivity: (wwan0,IPv4,37) check completed: FAKE; fake result
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9099] connectivity: (eth0,IPv6,24) check completed: FAKE; fake result
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9096] connectivity: (eth0,IPv4,23) check completed: FAKE; fake result
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9090] connectivity: (wwan0,IPv6,38) start fake request (fake result)
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9090] connectivity: (wwan0,IPv4,37) start fake request (fake result)
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9084] connectivity: (eth0,IPv6,24) start fake request (fake result)
Jan 11 11:21:21 27eb737 NetworkManager[1394]: <debug> [1673436081.9084] connectivity: (eth0,IPv4,23) start fake request (fake result)
```

As balenaOS 2.108.4 includes NetworkManager 1.40.4 (see https://github.com/balena-os/meta-balena/pull/2906), these logs come from https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/blob/1.40.4/src/core/nm-connectivity.c#L1134. Reading the lines above that, I think this is caused by the `WITH_CONCHECK` build flag not being present in the NetworkManager build included in balenaOS.

everhardt, 2023-01-11:
To elaborate: we want to use wwan0 as a failover for eth0. That also did not work in the previous balenaOS release we used (balenaOS 2.98.33), because that one contains a NetworkManager build that suffers from a bug causing hostname resolving to fail on failover connections in case systemd-resolved is not available (which is the case here). See https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/commit/57d226d3f08d8a904a554367e799c9c367032b0d for more detail.

majorz, 2023-01-11:
Thanks for the report and digging into this! As we upgraded to NetworkManager 1.40.4 we had to change the build system to meson and it looks like we missed this option. I will look into this with priority.

commit a7a1cd87 (2023-01-12): Enable back connectivity check in NetworkManager

In v2.108.0 we did a major NetworkManager upgrade as we started using
meson as a build system. The connectivity check feature was left
disabled, which is a regression and it produces fake results.

This commit enables it back.

Fixes #2964.

Change-type: patch
commit 95efb3f6 (2023-01-12): networkmanager: enable back connectivity check

In v2.108.0 we did a major NetworkManager upgrade as we started using
meson as a build system. The connectivity check feature was left
disabled, which is a regression and it produces fake results.

This commit enables it back.

Fixes #2964.

Changelog-entry: Enable back connectivity check in NetworkManager
Change-type: patch
commit 02724aae (2023-01-12): networkmanager: enable back connectivity check

In v2.108.0 we did a major NetworkManager upgrade as we started using
meson as a build system. The connectivity check feature was left
disabled, which is a regression and it produces fake results.

This commit enables it back.

Fixes #2964.

Changelog-entry: Enable back connectivity check in NetworkManager
Change-type: patch
commit e5ff0fe0 (2023-01-13): networkmanager: enable back connectivity check

In v2.108.0 we did a major NetworkManager upgrade as we started using
meson as a build system. The connectivity check feature was left
disabled, which is a regression and it produces fake results.

This commit enables it back.

Fixes #2964.

Changelog-entry: Enable back connectivity check in NetworkManager
Change-type: patch
closed 2023-01-20 by balena-ci