balena-os/meta-balena #2977: generic device type tests - DUT fails to be reachable

rcooke-warwick, 2023-01-18:
```
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:08:43.332Z][worker-cloud] GET http://worker/dut/ip
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:08:58.337Z][worker-cloud] GET http://worker/dut/ip
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:09:13.356Z][worker-cloud] GET http://worker/dut/ip
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:09:28.383Z][worker-cloud] # Subtest: Managed BalenaOS release suite
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:09:28.384Z][worker-cloud]     not ok 1 - 500 - "Could not resolve 3817e8f.local"
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:09:28.384Z][worker-cloud]       ---
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:09:28.384Z][worker-cloud]       stack: >
jenkins-leviathan-v2-template-11771-client-1  | [2023-01-11T20:09:28.385Z][worker-cloud]         new StatusCodeError (node_modules/request-promise-core/lib/errors.js:32:15)
```

seems to happen randomly right now, no cause identified

rcooke-warwick, 2023-01-18:
@vipulgupta2048 @rcooke-warwick will try to reproduce this locally as a first step

vipulgupta2048, 2023-01-26:
Hey @rcooke-warwick reproducing this now, the setup is a generic-amd64 image (latest) failing cloud test no?