balena-os/meta-balena #2998: HUP from 2.108.31 fails with out of space error on the boot partition

floion, 2023-02-02:
Checking the hup logs:

```
[INFO] Checking if boot partition can accommodate the new update... success.
[INFO] Updating timestamps for all files in /mnt/boot... done
[INFO] Deploying /mnt/boot/u-boot-fst-controller-dart.img... done.
[INFO] Deploying /mnt/boot/balena-image... done.
[INFO] Deploying /mnt/boot/SPL... done.
[INFO] Deploying /mnt/boot/image-version-info... done.
[INFO] Deploying /mnt/boot/zImage...Error: Failed to copy to a temporary location /mnt/boot/.zImage.bf5c18df.b96bdbfece22b30cb797b40ae8a3a663.md5sum -> /mnt/boot/.zImage.bf5c18df.b96bdbfece22b30cb797b40ae8a3a663.tmp

Caused by:
    No space left on device (os error 28)

Failed to run the new hooks. Running current hooks... 
[INFO] Checking if boot partition can accommodate the new update... fail.
[ERROR] Boot files copy operations will fail with out of space error.
```

alexgg, 2023-02-02:
@floion what device type is this?

alexgg, 2023-02-02:
@majorz I see what you say, but I think the copying function is the same for all boot files - and some files we do want to use fatrw on when updating - specifically `config.json`.  Could `fatrw` fallback to a normal copy if there is no space for a temp file to be created? In other ways, flag the error, but handle it?

majorz, 2023-02-02:
@alexgg nice idea. I will look into it - this should definitely be doable.

commit 585a5208 (2023-04-21): hostapp-update-hooks: use unsafe fatrw copy for bootfiles

A safe copy would only work for files that are read by fatrw also and that is
not the case for boot files. Still, some file like `config.json` would
benefit from a safe copy so we still try that first.

Fixes #2998

Change-type: patch
commit 94f8e80a (2023-04-21): hostapp-update-hooks: use unsafe fatrw copy for bootfiles

A safe copy would only work for files that are read by fatrw also and that is
not the case for boot files. Still, some file like `config.json` would
benefit from a safe copy so we still try that first.

Fixes #2998

Change-type: patch
commit 1e30c2c9 (2023-04-26): hostapp-update-hooks: use unsafe fatrw copy for bootfiles

A safe copy would only work for files that are read by fatrw also and that is
not the case for boot files. Still, some file like `config.json` would
benefit from a safe copy so we still try that first.

Fixes #2998

Change-type: patch
commit 71d1fa57 (2023-04-27): hostapp-update-hooks: use unsafe fatrw copy for bootfiles

A safe copy would only work for files that are read by fatrw also and that is
not the case for boot files. Still, some file like `config.json` would
benefit from a safe copy so we still try that first.

Fixes #2998

Change-type: patch
closed 2023-04-27 by alexgg