balena-os/meta-balena #3035: Capturing Docker Build Tag fails on Docker >23.0

kendallgoto, 2023-02-16:
[Starting in Docker 23.0.0 (2023-02-01)](https://github.com/docker/cli/blob/master/docs/deprecated.md#legacy-builder-for-linux-images), `docker build` was moved to alias as `docker buildkit build`. The buildkit builder doesn't output the same data onto stdout as the previous builder, which causes certain recipes reliant on `docker build` output to fail, such as:
https://github.com/balena-os/meta-balena/blob/e9e5601e6bb0c4ee4ab51094714f88ce074b4b09/meta-balena-common/recipes-containers/mkfs-hostapp-native/mkfs-hostapp-native.bb#L39-L41
Since there is no more stdout, IMAGE_ID has an empty value that is passed to `docker save`, creating the error
```
Error response from daemon: invalid reference format
```
[The envvar](https://github.com/docker/cli/blob/master/docs/deprecated.md#legacy-builder-fallback) `DOCKER_BUILDKIT=0` allows the original behavior to be restored, though it would be more appropriate to adjust the behavior to be compatible with the buildkit builder in the long run.

I was able to just add the environment variable here to resolve my build:
https://github.com/balena-os/meta-balena/blob/e9e5601e6bb0c4ee4ab51094714f88ce074b4b09/meta-balena-common/recipes-containers/mkfs-hostapp-native/mkfs-hostapp-native.bb#L39
https://github.com/balena-os/meta-balena/blob/e9e5601e6bb0c4ee4ab51094714f88ce074b4b09/meta-balena-common/recipes-kernel/linux/kernel-headers-test.bb#L40

But I'm not sure if this change also impacts other recipes that I wasn't reliant on.

alexgg, 2023-02-23:
hi @kendallgoto thanks for reporting this. The change probably needs to detect the docker version and act accordingly. meta-balena is currently based on Kirkstone, and I don't think the recommended Linux distributions for Kirkstone have updated to 23.0 yet (please correct me if I'm wrong). We will probably tackle this as part of the next Yocto version update.
As a side note there is a `balena-build.sh` script in `balena-yocto-script` that will build in a container and should isolate the build from this problems.

dadaroce, 2023-04-11:
Hey, I just noticed this issue, what about this solution? 

```.bash
    IMAGE_TAG="mkfs-hostapp-image"
    DOCKER_API_VERSION=1.22 docker build -t $IMAGE_TAG ${B}/work
    DOCKER_API_VERSION=1.22 docker save "$IMAGE_TAG" > ${B}/work/mkfs-hostapp-image.tar
    DOCKER_API_VERSION=1.22 docker rmi "$IMAGE_TAG"

    sed -i "s/@IMAGE@/${IMAGE_TAG}/" ${B}/work/mkfs.hostapp
```

I mean, we could just tag the Image with a defined name and avoid capturing the IMAGE_ID.

commit 79bccc93 (2023-06-16): mkfs-hostapp-native: Use image tags instead of parsing docker output

From docker 23.0 `docker build` is an alias to `docker buildkit build`
and the output is not parseable in the same way.

To avoid using unstable outputs as API, use image tags instead.

Fixes #3035

Change-type: patch
commit a27607bb (2023-06-19): mkfs-hostapp-native: Use image tags instead of parsing docker output

From docker 23.0 `docker build` is an alias to `docker buildkit build`
and the output is not parseable in the same way.

To avoid using unstable outputs as API, use image tags instead.

Fixes #3035

Change-type: patch
commit 99a8387a (2023-06-21): mkfs-hostapp-native: Use image tags instead of parsing docker output

From docker 23.0 `docker build` is an alias to `docker buildkit build`
and the output is not parseable in the same way.

To avoid using unstable outputs as API, use image tags instead.

Fixes #3035

Change-type: patch
commit a801afcf (2023-06-21): mkfs-hostapp-native: Use image tags instead of parsing docker output

From docker 23.0 `docker build` is an alias to `docker buildkit build`
and the output is not parseable in the same way.

To avoid using unstable outputs as API, use image tags instead.

Fixes #3035

Change-type: patch
closed 2023-07-04 by alexgg