balena-os/meta-balena #3101: The rootfs size 36864(K) exceeds IMAGE_ROOTFS_MAXSIZE: 32768(K)

RockSupport, 2023-04-08:
meta-balena/meta-balena-common/recipes-core/images/balena-image-initramfs.bb

New Module
    initramfs-module-recovery \
    initramfs-module-migrate \

The rootfs size 36864(K) exceeds IMAGE_ROOTFS_MAXSIZE: 32768(K)

RockSupport, 2023-04-09:
Need improve gcc Optimization level to -Os

alexgg, 2023-05-04:
There are a couple of solutions for the error above. One is to remove the extra modules from the initramfs if the device type does not need to support that functionality. For example:

* os-module-recovery is only needed for BSP development/debugging on production devices that don't have a serial port accessible 
* os-module-migrate is only required for devices that want to support the installer's migrator

Another possibility for device types that store the kernel and initramfs in the root partition is to just extend `IMAGE_ROOTFS_MAXSIZE`.

closed 2023-05-04 by alexgg