balena-os/meta-balena #3193: Moving to redsocks2 / shadowsocks?

shaunco, 2023-07-17:
Looking through issues, there seem to be several related to redsocks, that are either still open or were closed without fixes, including:
- #1522
- #1915
- #2021
- #2352 _(seems to be moving towards being IPv6 related)_
- #2408 

[redsocks](https://github.com/darkk/redsocks) has not been touched in 4 years. It really seems like time to move to [redsocks2](https://github.com/semigodking/redsocks) or [shadowsocks](https://github.com/shadowsocks/shadowsocks-rust) ... or best yet, redsocks2+shadowsocks, which provides much more modern support for the original goal of CloudConnect making it through a proxy while also supporting IPv6, enhanced encryption, and more proxy types.

kassett, 2024-12-05:
Is there any progress on this issue?

shaunco, 2026-01-01:
Thanks @ycardaillac, awesome to hear! Since I opened this issue, IETF has published RFC 9298 (CONNECT-UDP) and RFC 9484 (CONNECT-IP) covering [MASQUE](https://blog.cloudflare.com/zero-trust-warp-with-a-masque/). If the project picking process is still getting started, it would seem that MASQUE should be on the candidate board as well.