balena-os/meta-balena #3201: ModemManager does not allow sending AT commands over dbus in recent balenaOS version (~ 3.0.15)

everhardt, 2023-07-25:
I moved from balenaOS 2.108.29 to 3.0.15 and I noticed that I cannot query the modem anymore as I used to:

```
root@dd8d378:~# mmcli -m 0 --command=AT+CSQ
error: command failed: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: Cannot send AT command to modem: operation only allowed in debug mode'
```

We use this command over dbus from our container to monitor signal quality and read out other properties.

majorz, 2023-07-25:
And this option is missing from the recent transition to `meson` from `autotools`. It has to be added back.

pipex, 2023-08-21:
Another report in the link above about this also failing for Raspberry PI 4

majorz, 2023-08-22:
Related PR is https://github.com/balena-os/meta-balena/pull/3206

lmbarros, 2023-10-11:
Hi @majorz ! Can we close this issue now that the PR is merged?

everhardt, 2023-10-18:
We just tried this on balenaOS 4.0.26 and supervisor 14.13.4 but got the same error: `Cannot send AT command to modem: operation only allowed in debug mode`

alexgg, 2023-11-17:
Closed with https://github.com/balena-os/meta-balena/pull/3206

closed 2023-11-17 by alexgg

everhardt, 2023-11-20:
@alexgg We verified on 4.0.26 _and_ on 4.1.5 that this is not yet working.

everhardt, 2023-11-20:
Compulab IOT-GATE-iMX8.

floion, 2023-11-20:
Should be fixed by https://github.com/balena-os/meta-balena/pull/3298

floion, 2023-11-20:
@majorz can you give this a test on a rpi4 for example?

majorz, 2023-11-21:
@floion checking it now!

acostach, 2023-11-23:
Hi @everhardt, the [v4.1.5+rev2](https://github.com/balena-os/balena-iot-gate-imx8/commit/b9e127ae6e9eca09ff79618cd3b627f39eb4f981) iot-gate-imx8 image from balena-cloud should have this fixed, can you please give it a try and let us know?

everhardt, 2023-11-24:
Yes, that works, many thanks!

acostach, 2023-11-24:
Excellent, thanks for confirming @everhardt !

jdbruijn, 2024-01-03:
Hi @acostach, I'm a colleague of @everhardt and just tested this with the IOT-gate iMX8 with balenaOS 5.0.11 and supervisor 15.0.4 and unfortunately this seems broken again. We once again get the `Cannot send AT command to modem: operation only allowed in debug mode` error.

jdbruijn, 2024-01-03:
Apologies, disregard that message. I was testing on a device I didn't yet update. I can confirm it still works on balenaOS 5.0.11 and supervisor 15.0.4