balena-os/meta-balena #3205: Enabling ipv6 support in balenaEngine

majorz, 2023-07-28:
This is being marked as experimental feature in Docker, but we are already enabling that flag. The official documentation on enabling ipv6 is [here](https://docs.docker.com/config/daemon/ipv6/).

majorz, 2023-07-28:
I got an ipv6 address to be registered with:

```
version: '2.4'

services:
  hello-world:
    build: ./hello-world

    networks:
      customnet:
        ipv4_address: 172.28.0.2
        ipv6_address: 2001:3984:3989::20

networks:
  customnet:
    enable_ipv6: true
    ipam:
      driver: default
      config:
      - subnet: 172.28.0.0/14
        gateway: 172.28.0.1
      - subnet: 2001:3984:3989::/64
        gateway: 2001:3984:3989::1
```

The ipv4 subnet, gateway and addresses were needed as well because supervisor seems to reject the network without an ipv4 address.

ycardaillac, 2025-12-23:
Hello,

We are currently in the process of cleaning up our issues.

This one is (hopefully) not an issue anymore.

Bests,

Yann